zwave-js/zwave-js #5336: Fibaro FGK101 multilevel sensor disappears after restart

ScratMan, 2023-01-16:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

8.6.3

### ZwaveJS version

10.3.1

### Describe the bug

Hello, when I restart the ZWaveJS2MQTT docker container, the "Multilevel sensor" category of my Fibaro FGK-101 and the attached temperature sensor disappears from UI, and the corresponding sensor becomes unavailable in HA.
I then need to wait for the next wake up of the device for the temperature to reappear and comes back live in HA.

This started with an old release, maybe in V6, I could not identify which one exactly. I thought re-interviewing the device could solve the issue, but it didn't.

Before restart:
[image]

After restart:
[image]

### To Reproduce

Pair a FGK-101 with thermal sensor attached, get all sensors live and restart container.

### Expected behavior

Temperature sensor should be restored with last known value till wake up and value refresh from device.

ScratMan, 2023-01-23:
```
2023-01-23T17:52:49.757Z DRIVER   ███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗
                                  ╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝
                                    ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗
                                   ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║
                                  ███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║
                                  ╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝
2023-01-23T17:52:49.760Z DRIVER   version 10.3.1
2023-01-23T17:52:49.761Z DRIVER   
2023-01-23T17:52:49.762Z DRIVER   starting driver...
2023-01-23T17:52:49.798Z DRIVER   opening serial port /dev/serial/by-id/usb-0658_0200-if00
2023-01-23T17:52:49.816Z DRIVER   serial port opened
2023-01-23T17:52:49.818Z SERIAL » [NAK]                                                                   (0x15)
2023-01-23T17:52:49.831Z DRIVER   loading configuration...
2023-01-23T17:52:49.850Z CONFIG   Using external configuration dir /usr/src/app/store/.config-db
2023-01-23T17:52:49.851Z CONFIG   version 10.3.1
2023-01-23T17:52:51.125Z DRIVER   beginning interview...
2023-01-23T17:52:51.128Z DRIVER   added request handler for AddNodeToNetwork (0x4a)...
                                  1 registered
2023-01-23T17:52:51.129Z DRIVER   added request handler for RemoveNodeFromNetwork (0x4b)...
                                  1 registered
2023-01-23T17:52:51.130Z DRIVER   added request handler for ReplaceFailedNode (0x63)...
                                  1 registered
2023-01-23T17:52:51.131Z CNTRLR   querying controller IDs...
2023-01-23T17:52:51.392Z SERIAL » 0x01030020dc                                                         (5 bytes)
2023-01-23T17:52:51.395Z DRIVER » [REQ] [GetControllerId]
2023-01-23T17:52:51.402Z SERIAL « [ACK]                                                                   (0x06)
2023-01-23T17:52:51.410Z SERIAL « 0x01080120f176a1be014f                                              (10 bytes)
2023-01-23T17:52:51.413Z SERIAL » [ACK]                                                                   (0x06)
2023-01-23T17:52:51.417Z DRIVER « [RES] [GetControllerId]
                                    home ID:     0xf176a1be
                                    own node ID: 1
2023-01-23T17:52:51.438Z CNTRLR   received controller IDs:
                                    home ID:     0xf176a1be
                                    own node ID: 1
2023-01-23T17:52:51.438Z CNTRLR   querying API capabilities...
2023-01-23T17:52:51.453Z SERIAL » 0x01030007fb                                                         (5 bytes)
2023-01-23T17:52:51.454Z DRIVER » [REQ] [GetSerialApiCapabilities]
2023-01-23T17:52:51.457Z SERIAL « [ACK]                                                                   (0x06)
2023-01-23T17:52:51.461Z SERIAL « 0x012b0107010100860001005afe81ff884f1f0000fb9f7da067000080008086000 (45 bytes)
                                  000e87300000e0000600000fa
2023-01-23T17:52:51.463Z SERIAL » [ACK]                                                                   (0x06)
2023-01-23T17:52:51.466Z DRIVER « [RES] [GetSerialApiCapabilities]
                                    payload: 0x010100860001005afe81ff884f1f0000fb9f7da067000080008086000000e8730
                                  0000e0000600000
2023-01-23T17:52:51.474Z CNTRLR   received API capabilities:
                                    firmware version:    1.1
                                    manufacturer ID:     0x86
                                    product type:        0x01
                                    product ID:          0x5a
                                    supported functions: 
                                    · GetSerialApiInitData (0x02)
                                    · SetApplicationNodeInformation (0x03)
                                    · ApplicationCommand (0x04)
                                    · GetControllerCapabilities (0x05)
                                    · SetSerialApiTimeouts (0x06)
                                    · GetSerialApiCapabilities (0x07)
                                    · SoftReset (0x08)
                                    · GetProtocolVersion (0x09)
                                    · SetRFReceiveMode (0x10)
                                    · UNKNOWN_FUNC_SET_SLEEP_MODE (0x11)
                                    · FUNC_ID_ZW_SEND_NODE_INFORMATION (0x12)
                                    · SendData (0x13)
                                    · SendDataMulticast (0x14)
                                    · GetControllerVersion (0x15)
                                    · SendDataAbort (0x16)
                                    · FUNC_ID_ZW_R_F_POWER_LEVEL_SET (0x17)
                                    · UNKNOWN_FUNC_SEND_DATA_META (0x18)
                                    · FUNC_ID_ZW_GET_RANDOM (0x1c)
                                    · GetControllerId (0x20)
[... 1743 lines cut ...]
                                    is application CC:       true
                                    should hide root values: true
2023-01-23T17:55:05.815Z CNTRLR   [Node 016] [translateValueEvent: value updated]
                                    commandClass: Multilevel Sensor
                                    endpoint:     8
[... cut ...]
                                    should hide root values: true
2023-01-23T17:55:05.818Z DRIVER « [Node 016] [REQ] [ApplicationCommand]
                                  └─[MultiChannelCCCommandEncapsulation]
                                    │ source:      8
                                    │ destination: 0
                                    └─[MultilevelSensorCCReport]
                                        type:  Air temperature
                                        scale: Celsius
                                        value: 2.4

```

ScratMan, 2023-01-23:
A second log, maybe more complete. Looks like enabling or disabling the driver log makes the multilevel sensor disappear. Hope the disappear and reappear are catched in the logs.
The failing node is number 013.

[zwavejs_2023-01-23.log (2).txt](https://github.com/zwave-js/node-zwave-js/files/10482620/zwavejs_2023-01-23.log.2.txt)

AlCalzone, 2023-01-30:
Although you mentioned that you paired the device with sensor attached, this smells like the device does not actually report in the interview that it has this sensor.

Please re-interview the device and post a log of that. Ideally delete the logfile beforehand, so there's not so much unnecessary stuff.

closed 2023-02-07 by AlCalzone

ScratMan, 2023-02-24:
After updating with release including #5424 , the issue initially persists, with temperature sensor still disappearing when the ZWaveJS was restarted. So I finally had time to exclude the device from the network, made a hard reset of the device, and included it again in ZWaveJS-UI. But after inclusion, the temperature sensor was missing :flushed:

I retried several times without success, the multilevel sensor category didn't appear after inclusion, even after several manual wake-ups or re-interviews. I had the setting to set the temperature change threshold, but no temperature displayed.
So I gave up and left the device included as this, without the temperature sensor ; and after a little moment, the multilevel sensor category appeared, with the temperature sensor displayed inside :neutral_face:
I then tried to restart ZWaveJS, and ... temperature sensor remained available :star_struck:

So I confirm **the fix is working fine, but requires to exclude and include again the device** to work properly, and the temperature will appear after the next first temperature sampling.

AlCalzone, 2023-02-24:
That shouldn't be necessary, unless the device was actually doing something strange.

ScratMan, 2023-02-24:
> That shouldn't be necessary, unless the device was actually doing something strange.

It's possible the device connection was malfunctioning. But it finally works as expected now. Thanks