zwave-js/zwave-js #5346: How to debug "DRIVER Dropping message with invalid payload"

ukd1, 2023-01-18:
I can see where it's being [logged in the code](https://github.com/zwave-js/node-zwave-js/blob/b5336c049595b31aa5d7ae189973e91e06a06815/packages/zwave-js/src/lib/driver/Driver.ts#L2875), but the log message doesn't let me debug it further. There doesn't seem to be an obvious way to get any more logging, at least reading that code.

I have thousands of lines of:

```
2023-01-18T18:50:12.049Z DRIVER dropping CC with invalid values (Reason: Unsupported meter scale W or corrupte
d data)
2023-01-18T18:50:12.206Z DRIVER dropping CC with invalid values (Reason: Unsupported meter scale W or corrupte
d data)
2023-01-18T18:50:12.739Z DRIVER dropping CC with invalid values (Reason: Unsupported meter scale W or corrupte
d data)
2023-01-18T18:50:12.803Z DRIVER dropping CC with invalid values (Reason: Unsupported meter scale A or corrupte
d data)
2023-01-18T18:50:12.832Z DRIVER dropping CC with invalid values (Reason: Unsupported meter scale A or corrupte
d data)
2023-01-18T18:50:12.896Z DRIVER dropping CC with invalid values (Reason: Unsupported meter scale W or corrupte
d data)
2023-01-18T18:50:12.986Z DRIVER dropping CC with invalid values (Reason: Unsupported meter scale W or corrupte
d data)
2023-01-18T18:50:13.223Z DRIVER dropping CC with invalid values (Reason: Unsupported meter scale W or corrupte
d data)
2023-01-18T18:50:13.429Z DRIVER dropping CC with invalid values (Reason: Unsupported meter scale A or corrupte
d data)
2023-01-18T18:50:13.492Z DRIVER dropping CC with invalid values (Reason: Unsupported meter scale W or corrupte
d data)
2023-01-18T18:50:13.573Z DRIVER dropping CC with invalid values (Reason: Unsupported meter scale W or corrupte
d data)
2023-01-18T18:50:13.657Z DRIVER dropping CC with invalid values (Reason: Unsupported meter scale A or corrupte
d data)
```

Versions:
```
zwave-js-ui: 8.7.0
zwave-js: 10.3.1
```

How do I debug this?

AlCalzone, 2023-01-18:
Have you tried re-interviewing the device yet? This happens if the meter scale a device uses is not in the list of scales it has reported support for during the interview.

We filter those out because you can end up with tons of nonsensical measurements (like water consumption in an electrical meter) when packets get scrambled on air.

ukd1, 2023-01-18:
@AlCalzone thanks for the reply. TBH, this was working with no warnings, until recently. Unfortunately I don't know exactly when.

Anyway - I've re-interviewed all devices...but, that didn't fix it. To be honest I don't know which device this actually is - how can I tell from the log message?

ukd1, 2023-01-18:
> I'll check tomorrow and let you know. I'll need to see a log of the re-interview then.

How do I do that?

AlCalzone, 2023-01-19:
It's node ID 9.

> How do I do that?

Does the [documentation](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs) help? Loglevel `debug` (just like the log you posted above) and attach it here as a file (drag & drop into the text field).
Make sure to capture the entire re-interview process.

ukd1, 2023-01-19:
Closing, as weirdly it fixed itself after re-interviewing just that node; even though I'd re-interviewed all nodes before.

So, @AlCalzone thanks for pointing me in the right direction - would love to how you figured out it was node `9` from the logs? I can't see it mentioned there? I saw `0x06` -- but that's just `6`...?

closed 2023-01-19 by ukd1

AlCalzone, 2023-01-19:
I guess you could figure out by knowing which byte to look at. There is a 09 in there btw.

However, I copied the serial message into my [decoder script](https://github.com/zwave-js/node-zwave-js/blob/master/test/debug.js) and executed it, then inspected the decoded message.