zwave-js/zwave-js #5347: Restoring NVM from Aeotec Z-Stick Gen5 to Aeotec Z-Stick Gen7 controller marks all nodes as dead

Kyoshi67, 2023-01-18:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

YES, BUT a Z-Wave JS UI developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Restoring the NVM backup from a Aeotec Gen5 Z-Stick to a Aeotec Gen7 Z-Stick. 

What do you observe?
All nodes are dead.

What did you expect to happen?
I expected all the nodes on my Aeotec Gen5 to be functional on my Aeotec Gen7.

Steps to reproduce the behavior:
  1. Backed up my Aeotec Gen5 with Z-Wave JS UI
  2. Restore the backup of my Gen5 on my Gen7 with Z-Wave JS UI
  3. Restart

### Device information

Old controller : Aeotec Gen5 SDK 6.7
New controller : Aeotec Gen7 SDK 7.18.1
Home automation system : Jeedom with Z-Wage JS plugin

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [X] Other _(please describe)_

### Which branches or versions?

Jeedom 4.3.14 with Z-Wage JS plugin 2022-12-25 01:03:08 (Wave JS UI 8.6.1).

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwavejs_2023-01-19.log](https://github.com/zwave-js/node-zwave-js/files/10451938/zwavejs_2023-01-19.log)

AlCalzone, 2023-02-16:
I believe this should be fixed by the changes I did in `v10.10.0`: https://github.com/zwave-js/node-zwave-js/releases/tag/v10.10.0
Please update to the latest version, then try to restore the NVM on the 700 series stick again.

Kyoshi67, 2023-02-16:
> I believe this should be fixed by the changes I did in `v10.10.0`: https://github.com/zwave-js/node-zwave-js/releases/tag/v10.10.0 Please update to the latest version, then try to restore the NVM on the 700 series stick again.

Yesssssss!!!!
Thank you very much! My Aeotec Gen7 has been sitting in a drawer for months.
I backed up and restored it to the latest version, put my Gen7 on my home automation server and everything seems to be working as it should.

closed 2023-02-16 by AlCalzone