zwave-js/zwave-js #5351: Zipato KeyPad switch don't work

max3163, 2023-01-22:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

The Switch is never updated when I arm or disarm my ZIPATO keypad device. 
Nothing is updated on any switch remote to my Domoticz system too.

In fact, when I switch home or away in the tag reader just an event/notification is fire :

```
2023-01-14T19:58:07.764Z CNTRLR   [Node 006] [Notification]
type:   Access Control
event:  Keypad unlock operation
userId: 4
```

But the switch remain not updated...
```
Binary Switch v1
[6-37-0-currentValue] Current value : false
```

Is-it possible to add/update a switch corresponding to the Access_Control event for Keypad_unlock_operation/Keypad_lock_operation event? 

On MQTT log : 
```
2023-01-14T19:58:07: Received PUBLISH from ZWAVE_GATEWAY-zwave-js-ui (d0, q1, r0, m12351, 'domoticz/_EVENTS/ZWAVE_GATEWAY-zwave-js-ui/node/node_value_updated', ... (956 bytes))
2023-01-14T19:58:07: Sending PUBLISH to Domoticz-MQTT-AutoDiscoverd96c4a0c-2a15-40d8-a202-c640693b508011 (d0, q0, r0, m0, 'domoticz/_EVENTS/ZWAVE_GATEWAY-zwave-js-ui/node/node_value_updated', ... (956 bytes))
2023-01-14T19:58:07: Sending PUBACK to ZWAVE_GATEWAY-zwave-js-ui (m12351, rc0)
2023-01-14T19:58:07: Received PUBLISH from ZWAVE_GATEWAY-zwave-js-ui (d0, q1, r0, m12352, 'domoticz/Entrée/Zipato_RFID_Clavier/113/0/Access_Control/Keypad_unlock_operation', ... (43 bytes))
```

### Device information

``` "productLabel": "MINI KEYPAD RFID",
  "productDescription": "Keypad with alarm activation rfid tag",
  "manufacturer": "Wintop",
  "firmwareVersion": "0.28",
  "protocolVersion": 3,
  "nodeType": 1,
  "endpointsCount": 0,
  "endpointIndizes": [],
  "isSecure": false,
  "security": "None",
  "supportsSecurity": false,
  "supportsBeaming": true,
  "isControllerNode": false,
  "isListening": false,
  "isFrequentListening": false,
  "isRouting": true,
  "keepAwake": false,
  "maxDataRate": 40000,
  "deviceClass": {
    "basic": 4,
    "generic": 64,
    "specific": 0
      "id": "6-37-0-currentValue",
      "nodeId": 6,
      "toUpdate": false,
      "commandClass": 37,
      "commandClassName": "Binary Switch",
      "endpoint": 0,
      "property": "currentValue",
      "propertyName": "currentValue",
      "type": "boolean",
      "readable": true,
      "writeable": false,
      "label": "Current value",
      "stateless": false,
      "commandClassVersion": 1,
      "list": false,
      "value": false,
      "isCurrentValue": true,
      "targetValue": "37-0-targetValue",
      "lastUpdate": 1673777986262,
      "newValue": false
    ```

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

2023-01-22T10:39:28.782Z DRIVER   version 10.3.1
2023-01-22T10:39:28.783Z DRIVER   
2023-01-22T10:39:28.805Z CONFIG   version 10.3.1

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### If yes, where did it work?

When I used the OpenZwave lib with domoticz it worked well.

### Attach Driver Logfile

[zwavejs_2023-01-22.log](https://github.com/zwave-js/node-zwave-js/files/10474193/zwavejs_2023-01-22.log)

AlCalzone, 2023-01-30:
Assuming [this](https://www.zipato.com/wp-content/uploads/2015/07/wt-rfid-Zipato-Mini-Keypad-RFiD-Z-Wave-User-Manual-v1.4.pdf) is your device manual, then the switch you reference is this one:
> The Switch Binary Command Class is used to enable or disable
the notification sound. This sound is typically used to notify a user
when the alarm system is being activated. See also the ‘Sound
Notification’ section.

which has nothing to do with what you're trying to do.

Keypad unlock notifications should be mapped to values of either the Lock CC or the Door Lock CC. Unfortunately, I cannot see this in your log, since it is not on loglevel **debug** as requested.

AlCalzone, 2023-01-30:
Uhhhh - there's no (Door) Lock CC to map these events to.

So you'll have to implement something with MQTT discovery, but that's not a driver issue, but zwave-js UI. I'll transfer the issue. @robertsLando should be able to give you some pointers.