zwave-js/zwave-js #5352: Heltun FT01 and HT01 no longer synchronizes clock

johanschelin, 2023-01-22:
In the beginning (like 6m+ ago) this worked fine. Am not sure when it changed, could be v10, or a newer firmware in the FT/HT01s.

The issue is that the clock no longer synchronizes. Tried to trigger a sync via changing parameter 19 "Time correction by controller", and starting debug logging and this is what I get. Is this something that could be an issue in ZWaveJS, or is it in the firmware of the device you think? 

2023-01-22T11:02:18.977Z SERIAL « 0x010d00040098056c0232ff00c3000b (15 bytes)
2023-01-22T11:02:18.979Z SERIAL » [ACK] (0x06)
2023-01-22T11:02:18.982Z DRIVER « [Node 152] [REQ] [ApplicationCommand]
└─[SupervisionCCReport]
session id: 50
more updates follow: false
status: Success
duration: 0s
2023-01-22 12:02:18.987 INFO Z-WAVE: Node 152: value updated: 112-0-19 0 => 1
2023-01-22 12:02:18.988 INFO Z-WAVE: Success zwave api call writeValue true
2023-01-22T11:02:23.962Z SERIAL « 0x010c00040098048106ec0cc600ca (14 bytes)
2023-01-22T11:02:23.967Z SERIAL » [ACK] (0x06)
2023-01-22T11:02:23.970Z DRIVER « [Node 152] [REQ] [ApplicationCommand]
└─[ClockCCReport]
clock setting: Sunday, 12:12
2023-01-22T11:02:23.973Z CNTRLR [Node 152] detected a deviation of the node's clock, updating it...
2023-01-22T11:02:23.986Z SERIAL » 0x010f001398086c01b3048104ec022501e6 (17 bytes)
2023-01-22T11:02:23.987Z DRIVER » [Node 152] [REQ] [SendData]
│ transmit options: 0x25
│ callback id: 1
└─[SupervisionCCGet]
│ session id: 51
│ request updates: true
└─[ClockCCSet]
clock setting: Sunday, 12:02
2023-01-22T11:02:23.990Z SERIAL « [ACK] (0x06)
2023-01-22T11:02:23.994Z SERIAL « 0x0104011301e8 (6 bytes)
2023-01-22T11:02:23.995Z SERIAL » [ACK] (0x06)
2023-01-22T11:02:23.996Z DRIVER « [RES] [SendData]
was sent: true
2023-01-22T11:02:24.013Z SERIAL « 0x011800130100000200c67f7f7f7f000003000000000301000030 (26 bytes)
2023-01-22T11:02:24.013Z SERIAL » [ACK] (0x06)
2023-01-22T11:02:24.014Z DRIVER « [REQ] [SendData]
callback id: 1
transmit status: OK, took 20 ms
routing attempts: 1
protocol & route speed: Z-Wave, 100 kbit/s
ACK RSSI: -58 dBm
ACK channel no.: 0
TX channel no.: 0
2023-01-22T11:02:24.025Z SERIAL « 0x010d00040098056c02330200c600f2 (15 bytes)
2023-01-22T11:02:24.026Z SERIAL » [ACK] (0x06)
2023-01-22T11:02:24.026Z DRIVER « [Node 152] [REQ] [ApplicationCommand]
└─[SupervisionCCReport]
session id: 51
more updates follow: false
status: Fail
duration: 0s

AlCalzone, 2023-01-22:
It seems like those devices don't like that the driver now uses Supervision CC for Set-type commands.

johanschelin, 2023-01-22:
Ok. Is that something we can work around or does it need to fixed by Heltun?

Skickat från Outlook för iOS<https://aka.ms/o0ukef>

AlCalzone, 2023-01-22:
I've reached out to them. An easy workaround would be disabling Supervision for those devices in our config files, but we actually need this for S2 to work reliably (which wouldn't be an issue in your specific setup though).

johanschelin, 2023-01-23:
Hi!

I had also already e-mailed the support and this is actually fixed in a later firmware (2.6+)
I have received the newest firmware file from them and tested, and can confirm that this has solved the issue!

This issue can now be closed, letting you close it so you have time to see my answer.

I have asked them to publish the new firmware as the latest on their supportsite is 2.5 which does not include the Supervised Clock fix.

/Schelin

closed 2023-01-25 by AlCalzone