zwave-js/zwave-js #5354: Honeywell T6 Pro Z-Wave (TH6320ZW2007)

ufgrat, 2023-01-23:
There is a new version of the T6 Pro Z-Wave thermostat which is not recognized by the Z-Wave JS UI.  The old model is the TH6320ZW2003 (supported) and the new model is the TH6320ZW2007.

Due to internal restructuring at Honeywell, the manufacturer ID has changed from Honeywell (0x0039) to Resideo (0x041B).  Similarly, the device ID has changed from 0x0011:0x0008 to 0x0011:0x0009.  It now supports Z-Wave+ / S2.

The result is that while Z-Wave JS UI under Home Assistant discovers the device reasonably correctly, it doesn't report the parameter names / values correctly, and specifically I'm having a difficulty with the "mode" setting-- but unless I have a working device config, I can't guarantee there isn't a misconfiguration somewhere.

I would be happy to work with you to get the device file working (There are many, MANY unlabeled parameters right now).  I'd even take a shot at it myself, but I was unable to decipher the location of the existing device config files for someone who's never developed with Z-Wave JS UI before.

AlCalzone, 2023-01-30:
fixed in #5143

closed 2023-01-30 by AlCalzone

PaddyDHurley, 2026-01-31:
@AlCalzone I am having an issue where if I turn the thermostat on at the panel, it doesn’t always update the ios app mode.  Others have reported an earlier version of firmware on their device.  Mine is 3.1.0.  Can i work with you on this issue?

[image]

AlCalzone, 2026-02-03:
Please create a new discussion [here](https://github.com/zwave-js/zwave-js/discussions/new?category=request-support-investigate-issue)