zwave-js/zwave-js #5356: OTW update of Zooz ZST39 800-series controller stick completed successfully with timeout error.

kdknigga, 2023-01-23:
Updated my ZST39 from firmware 1.1 to 1.2 using ZUI 8.8.0.  The update attempt ended with a timeout error.

```
2023-01-23T18:24:21.080Z SERIAL » 0x011be47b862868138c2b549e98486e7b6c5a8df82e144134869db8117e96c92e (133 bytes)                                                                                                   
                                  3086d3462ac65a5f76dafc0404fc04000000b4eccf06ffffffffffffffffffffffffffffffffff                                                                                                   
                                  ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff                                                                                                   
                                  ffffffffffffffffffffffffffffffffffffffffff7549                                                                                                                                   
2023-01-23T18:24:21.102Z SERIAL « 0x06                                                                 (1 bytes)                                                                                                   
2023-01-23T18:24:21.103Z SERIAL » 0x04                                                                 (1 bytes)                                                                                                   
2023-01-23T18:24:21.105Z SERIAL « 0x06                                                                 (1 bytes)                                                                                                   
2023-01-23T18:24:21.123Z CNTRLR   [BOOTLOADER] Serial upload complete                                                                                                                                              
2023-01-23T18:24:22.124Z CNTRLR   OTW update failed: The bootloader did not acknowledge the end of transfer.                                                                                                       
2023-01-23 18:24:22.126 INFO Z-WAVE: Controller 1 firmware update OTW finished with error.                                                                                                                         
   Status: Error_Timeout. Result: [object Object].                                                                                                                                                                 
2023-01-23T18:24:22.126Z CNTRLR   Leaving bootloader...                                                                                                                                                            
2023-01-23T18:24:22.127Z SERIAL » 0x32                                                                 (1 bytes)                                                                                                   
2023-01-23T18:24:22.127Z CNTRLR   Restarting driver after OTW update...
```

Despite the error, after ZUI restarted the stick reported the new firmware version and works as expected.

andronat48, 2023-01-24:
Where did you get the firmware? I can see ZST39_SDK_7.18.3_US.gbl on the Zooz website, is it newer?

kdknigga, 2023-01-24:
@andronat48, yes, that firmware is newer than what shipped on my stick.

pyrodex, 2023-03-05:
I was running 7.18.1 which came with my 800 series stick, just upgraded to 7.18.3 via the firmware download from ZooZ. While it produced an error like @kdknigga 's experience the stick came back up running 7.18.3 and tested various devices.

btehan, 2023-09-22:
[image]
I got the same thing via HA.

AlCalzone, 2024-05-17:
Another log:
```
13:38:51.536 SERIAL « 0x06                                                                 (1 bytes)
13:38:51.537 SERIAL » 0x0149b6f661ebf00986bac6c67e7a00b3d2aed71286b545ad47bf32d1d5917e52 (133 bytes)
                      0f2ad55d3d0bb1e2578f5989a0c5044682f20a571bc221b66e49bb90ba849e885e325e7be87c43
                      421dea3403fd2d40de20118b9b05fc7b1210d1ab8c2142ec407911438ed5d15128988bd727ea72
                      c138f7a3f8415152abe315f70a0af7400000008ba05e78
13:38:51.569 SERIAL « 0x06                                                                 (1 bytes)
13:38:51.570 SERIAL » 0x014ab5951bfd9ddf65e8d65f9c7b5fb24556951d8ef9268669289859e087099b (133 bytes)
                      49d86a2bed0127f03c3325dcfd559f97abc7c636329692e611da9d7d144bd00fcafc0404fc0400
                      00004f79bdc1ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
                      ffffffffffffffffffffffffffffffffffffffffff22b1
13:38:51.592 SERIAL « 0x06                                                                 (1 bytes)
13:38:51.592 SERIAL » 0x04                                                                 (1 bytes)
13:38:51.594 SERIAL « 0x06                                                                 (1 bytes)
13:38:51.612 SERIAL « Serial upload complete
13:38:51.613 CNTRLR   [BOOTLOADER] Serial upload complete
13:38:51.613 SERIAL « Gecko Bootloader v2.01.02
                      1. upload gbl
                      2. run
                      3. ebl info
                      BL >
13:38:52.614 CNTRLR   OTW update failed: The bootloader did not acknowledge the end of transfer.
```

closed 2024-06-10 by zwave-js-bot