zwave-js/zwave-js #5360: Updating Aeotec S7 to 7.19.0 using Zwave JS UI disabled my controller

alexruffell, 2023-01-25:
### Checklist

- [ ]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

8.8.0

### ZwaveJS version

10.4.0

### Describe the bug

I initiated an upgrade of my Aeotec S7 using the new OTW functionality and soon after starting the update I got these errors:

[image]

Driver: Failed to recover from bootloader. Please flash a new firmware...

Problem is that neither HA or Zwave PC Controller see the stick as connected. It sees the COM port but not the Zwave stick so I cannot flash a new firmware or even restore the NVM I backed up just before testing this.

[image]

### To Reproduce

Update to 7.19.0 using Zwave JS UI OTW. The file I used was BRD4206A as it is the same board type as the one on Aeotec's site.

### Expected behavior

n/a

AlCalzone, 2023-01-25:
I waited for this to happen 🙈
Will make a small utility later today or tomorrow.

robertsLando, 2023-01-25:
Try to enable boot loader only mode under zwave settings and then trigger the fwupdate from there

alexruffell, 2023-01-25:
@robertsLando I can't get to the OTW Update function as nothing shows up in the list of devices:
[image]

robertsLando, 2023-01-25:
Shit 😅 the only way would be to do it using a driver function, the file should be placed on store dir and the function should read the file from there... @AlCalzone could you provide that function code? Otherwise I will do it tomorrow, not on my pc right now

alexruffell, 2023-01-25:
@robertsLando That may be an issue as well because ZWave JS UI is installed as an add-on to Home Assistant so I don't believe I have any access to the docker it is installed in.

I ordered a new s7 arriving tomorrow and am hoping I can restore the NVM to the new one to get back up and running. I'll need to research the exact process to avoid messing that one up too. Then I can set up some other system to restore this one if at all possible.

AlCalzone, 2023-01-25:
You can upload a file using the built in store browser, but I have no idea how to access it from a driver function.

alexruffell, 2023-01-25:
@AlCalzone - Is there a flasher tool I can use in a Linux VM or even better in Windows?

AlCalzone, 2023-01-25:
In an hour or so. Will ping you when it is done

alexruffell, 2023-01-25:
> In an hour or so. Will ping you when it is done

I restored the NVM to a Zooz 700 series stick I had and my network is back up. I will likely migrate t the 800 series stick when it arrives this weekend since it was insanely easy (Thank you devs!!!). When the tool is ready, I will try to restore the s7 stick, but there is no rush. Either way, thank you!

AlCalzone, 2023-01-25:
Unfortunately, migrating to 800 series won't be as easy, since the NVM backup/restore doesn't work there (yet?).
As for the flasher utility, this is now live:
https://github.com/zwave-js/node-zwave-js/tree/master/packages/flash#usage

For example:
```
npx @zwave-js/flash /dev/ttyUSB0 /path/to/firmware.otz
```

All you need is a somewhat up to date version of Node.js (prefer LTS): https://nodejs.org/en/

alexruffell, 2023-01-25:
I will need to learn how to use node.js :(

[image]

robertsLando, 2023-01-26:
@alexruffell YOu don't need to know how to use nodejs but just to install it and run the command @AlCalzone provided to you with the path to the file

AlCalzone, 2023-01-26:
yeah, Node.js just needs to be installed to have the `npx` executable. That is just a CLI command, you don't have to execute node yourself.

robertsLando, 2023-01-26:
Zwave-JS UI 8.8.1 will have the action on main advanced actions dialog so the problem would be fixed. Releasing it now...

alexruffell, 2023-01-26:
@AlCalzone As seen in the screenshot from yesterday, I figured out that the npx command needs to be run from the powershell and not within the node.js window. I don't know if my syntax is correct but the tool says "Flashing firmware..." and then stops. I left it all night as I had to leave my computer and it did not do anything further. I tried again this morning with the same result. I'll research this a bit more to verify syntax.

AlCalzone, 2023-01-26:
You didn't have to install `zwave-js` separately. Simply run the `npx ...` command.

To debug what's going on, you can add ` --verbose` to the end of the command. This will show the driver logs in the console.

alexruffell, 2023-01-26:
Ok, it is not recovering from bootloader:

[image]

AlCalzone, 2023-01-26:
Hmm so it's seeing the bootloader menu but apparently the flashing process does not start. Unfortunately that communication is hidden in the logs. I will see if I can show that too in verbose mode.

AlCalzone, 2023-01-26:
Ok new version is out. Please use this command to make sure you get the newest version:
```
npx @zwave-js/flash@latest ... --verbose
```
(replace ... with your port and filename)

alexruffell, 2023-01-26:
[image]

AlCalzone, 2023-01-26:
damn, that's not much...
let me try to replicate this situation - I already ended up there once

AlCalzone, 2023-01-26:
Ok, can reproduce :)
Next release should solve this.

AlCalzone, 2023-01-26:
Try again in 20 minutes or so when the release is done.

alexruffell, 2023-01-26:
[image]
[image]

AlCalzone, 2023-01-26:
Honestly I'm not sure what you gain with 800 series right now. If you really want to, you'll have to migrate the old fashioned way by excluding devices on the old controller and including them on the new one.

alexruffell, 2023-01-26:
I am always looking to improve range and network bandwidth to eliminate issues that have plagued me for years. I am just not clear on whether a subset of 800 series improvements can benefits older series as well, or whether *all* improvements require the nodes to be 800 series as well. I'll do my research... Thanks again!

alexruffell, 2023-01-26:
@AlCalzone  - Regarding the cause of this issue, could it be that HA was flooding the controller with zwave traffic during the OTW update? I recall seeing a timeout error before the bootloader one. If so, then would there be some mechanism to stop HA from sending traffic to the controller during an OTW update other than manually disabling the websocket?

AlCalzone, 2023-01-26:
The zwave-js driver pauses the send queue when doing OTW updates, since those messages can't be sent anyways while in the bootloader.

 Maybe Z-UI did something that caused an error to be thrown and killed the process, but I guess we won't know what unless you reproduce it while making log files.
The way I got my Stick stuck in the bootloader was by pulling it out during the update.

alexruffell, 2023-01-26:
@AlCalzone Turns out I was saving logs to file... 

[zwavejs_2023-01-25.log.gz](https://github.com/zwave-js/node-zwave-js/files/10513648/zwavejs_2023-01-25.log.gz)

While I can't make sense of it, I did find the spot where it failed the OTW. Hopefully it helps you troubleshoot what happened. Please remove the file, or let me know to do so, if it has info that should not be on here.

BTW - is having it log to file bogging things down?

AlCalzone, 2023-01-26:
Uhh that is indeed helpful. I'll leave the gist here for my later reference, then delete your file:

Firmware update starts in the middle of an API command
```
2023-01-25T15:23:19.912Z DRIVER » [Node 120] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x24
                                  │ callback id:      0
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 17
                                    └─[SupervisionCCReport]
                                        session id:          63
                                        more updates follow: false
                                        status:              Success
                                        duration:            0s
2023-01-25T15:23:19.919Z SERIAL « [ACK]                                                                   (0x06)
2023-01-25T15:23:19.919Z SERIAL « 0x010401a90152                                                       (6 bytes)
2023-01-25T15:23:19.920Z SERIAL » [ACK]                                                                   (0x06)
2023-01-25T15:23:19.920Z DRIVER « [RES] [SendDataBridge]
                                    was sent: true
2023-01-25T15:23:25.943Z SERIAL « 0x010c00a80001a103250300009a44                                      (14 bytes)
2023-01-25T15:23:25.944Z SERIAL » [ACK]                                                                   (0x06)
2023-01-25T15:23:25.945Z CNTRLR   [Node 161] [~] [Binary Switch] currentValue: false => false       [Endpoint 0]
2023-01-25T15:23:25.945Z DRIVER « [Node 161] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -102 dBm
                                  └─[BinarySwitchCCReport]
                                      current value: false
2023-01-25T15:23:29.982Z SERIAL « 0x010c00a80001a103250300009947                                      (14 bytes)
2023-01-25T15:23:29.983Z SERIAL » [ACK]                                                                   (0x06)
2023-01-25T15:23:29.985Z CNTRLR   [Node 161] [~] [Binary Switch] currentValue: false => false       [Endpoint 0]
2023-01-25T15:23:29.986Z DRIVER « [Node 161] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -103 dBm
                                  └─[BinarySwitchCCReport]
                                      current value: false
2023-01-25T15:23:34.058Z SERIAL « 0x010c00a80001a103250300009846                                      (14 bytes)
2023-01-25T15:23:34.059Z SERIAL » [ACK]                                                                   (0x06)
2023-01-25T15:23:34.059Z CNTRLR   [Node 161] [~] [Binary Switch] currentValue: false => false       [Endpoint 0]
2023-01-25T15:23:34.060Z DRIVER « [Node 161] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -104 dBm
                                  └─[BinarySwitchCCReport]
                                      current value: false
2023-01-25T15:23:41.376Z CNTRLR   Entering bootloader...
2023-01-25T15:23:41.376Z CNTRLR   Performing soft reset...
2023-01-25T15:23:41.379Z SERIAL » 0x01030008f4                                                         (5 bytes)
2023-01-25T15:23:41.380Z DRIVER » [REQ] [SoftReset]
2023-01-25T15:23:41.383Z SERIAL « [ACK]                                                                   (0x06)
2023-01-25T15:23:41.385Z CNTRLR   Waiting for the controller to reconnect...
2023-01-25T15:23:42.632Z SERIAL « 0x0119000a03000102010f5e86727322855970565a989f6c557400f3            (27 bytes)
2023-01-25T15:23:42.633Z SERIAL » [ACK]                                                                   (0x06)
2023-01-25T15:23:42.633Z DRIVER « [REQ] [SerialAPIStarted]
                                    wake up reason:        WatchdogReset
                                    watchdog enabled:      false
                                    generic device class:  0x02
                                    specific device class: 0x01
                                    always listening:      false
                                    supports Long Range:   false
2023-01-25T15:23:42.635Z CNTRLR   reconnected and restarted
2023-01-25T15:23:42.635Z CNTRLR   Enabling Smart Start listening mode...
2023-01-25T15:23:42.637Z SERIAL » 0x01030027db                                                         (5 bytes)
2023-01-25T15:23:42.638Z SERIAL » 0x0105004a4900f9                                                     (7 bytes)
2023-01-25T15:23:42.638Z DRIVER » [REQ] [AddNodeToNetwork]
                                    action: Enable Smart Start listening mode
2023-01-25T15:23:42.651Z SERIAL « 0x06                                                                 (1 bytes)
2023-01-25T15:23:42.653Z CNTRLR   [BOOTLOADER] version 1.7.1
2023-01-25T15:23:42.653Z CNTRLR   Entered bootloader
2023-01-25T15:23:42.653Z CNTRLR   Beginning firmware upload
2023-01-25T15:23:42.654Z SERIAL » 0x31                                                                 (1 bytes)
2023-01-25T15:23:42.658Z CNTRLR   [BOOTLOADER] begin upload
2023-01-25T15:23:42.663Z SERIAL « 0x43                                                                 (1 bytes)
2023-01-25T15:23:42.663Z CNTRLR   [BOOTLOADER] awaiting input...
[... 24 lines cut ...]
                                    action: Enable Smart Start listening mode
[... cut ...]
                                  n 1100 ms.
2023-01-25T15:23:44.773Z CNTRLR   OTW update failed: The bootloader did not acknowledge the start of transfer.
2023-01-25T15:23:44.773Z CNTRLR   Leaving bootloader...
2023-01-25T15:23:44.774Z SERIAL » 0x32                                                                 (1 bytes)
2023-01-25T15:23:44.774Z CNTRLR   Restarting driver after OTW update...
2023-01-25T15:23:44.774Z DRIVER   destroying driver instance...
2023-01-25T15:23:44.833Z DRIVER   driver instance destroyed
```

PR #5368 (2023-01-27): fix: prevent ongoing transactions from messing with the bootloader
fixes: #5360 

The main issue was that soft-resetting restarts the auto-provisioning mode. Since this request doesn't get an answer, it used to get re-transmitted while in bootloader mode - and worse, while transferring a firmware update.
We now reject all pending transactions twice (once before and once after pausing the send queue) to be extra sure. And we prevent the auto-provisioning when we're about to enter the bootloader.

There's probably a more elegant fix for this, but this would require reworking the state machines a [... cut]
closed 2023-01-27 by AlCalzone