zwave-js/zwave-js #5369: Error in log pointing to possible memory leak

johanschelin, 2023-01-29:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

I am not sure if this is important, but will log the issue to see, as I can't find any useful information from googling this:

What causes the bug?
Not sure

What do you observe?
Only that this error is written in the log.

Steps to reproduce the behavior:
As soon as I do a restart av the Add-on this always arises.

The error that is logged is the following:
(node:157) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 error listeners added to [Keyv]. Use emitter.setMaxListeners() to increase limit
(Use `node --trace-warnings ...` to show where the warning was created)

Is this something known that I should only disregard, or is it something you think we should dig deeper into?

### Device information

Manufacturer: 
Model name: 
Node ID in your network: 

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwave-js-ui: 8.8.0
zwave-js: 10.4.0
home id: 4110551221
home hex: 0xf50208b5

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

Will attach log if you think this issue is worth digging into. (Am in the middle of a long running firmware update, so dont want to restart and change loglevel now)

AlCalzone, 2023-01-29:
@robertsLando does **Keyv** ring a bell with you?

robertsLando, 2023-01-30:
@AlCalzone By checking yarn.lock it's a dependency of `cacheable-request` and `esm2cjs` modules, the first is also a dep of `got` that I think is what you are using to make some api calls, could be that maybe?

AlCalzone, 2023-01-30:
Looks I need to half-revert https://github.com/zwave-js/node-zwave-js/pull/5090

PR #5371 (2023-01-30): fix: use Keyv instance directly as got request cache
fixes: #5369
closed 2023-01-30 by AlCalzone