zwave-js/zwave-js #5372: An option to heal all powered nodes

ColColonCleaner, 2023-01-30:
**Is your feature request related to a problem? Please describe.**
When calling for a network heal it includes every node in the network. I don't want to go around to the 30 battery nodes every time so it can mark the heal as 'complete'. Whenever a heal is waiting on battery nodes to connect the heal can't be ended. The 'stop healing' option doesn't seem to do anything as the nodes still have their loading masks in the heal column after stopping the heal. And after stopping a network heal before it's marked 'complete' by going through even node, individual heals on nodes get automatically marked failed without trying. I know this happens when doing two individual heals at once, it looks like they can't be queued so i need to wait for one to finish before calling the next one but that's fine for ones and twos.

If this is just a bug where stopping the heal doesn't actually stop things and mark it complete then i can create a separate issue for that. The only way I've found to get it to stop fully is by rebooting the VM running HA.

**Describe the solution you'd like**
Regardless of whether it's a bug that led me down this line of thinking, having an option to heal all powered nodes so the user can try getting the backbone of their network solid before doing an all inclusive heal would be great.

**Describe alternatives you've considered**
I've currently been using the save settings feature on ZUI for something similar to this. I use it very frequently. Just going into settings and hitting save bounces the driver and calls for all of the powered nodes to communicate. Essentially using the raw communication from all those powered nodes as a way to heal things. Not sure if I'm shooting myself in the foot by doing it this way but I'm sure I'll find out shortly lol. Sometimes I'll do it many times in a row since it only takes about a minute or so for the communication to settle down and finish after the save.

**Additional context**

AlCalzone, 2023-01-30:
I need to double-check but this could just be a UI issue where Z-UI doesn't update the heal status when aborting the heal.

AlCalzone, 2023-01-30:
It is => https://github.com/zwave-js/zwave-js-ui/issues/2926

ColColonCleaner, 2023-01-31:
It would be nice to have the powered nodes option because if i start doing something else and forget to stop the heal once the powered nodes are done the network is essentially unusable. Tons of dead nodes and dropped commands if I forget to stop it lol.

ColColonCleaner, 2023-01-31:
Are you sure it actually stops the heal when stop is pressed? You said it successfully calls the stop command, but does the heal actually stop?

I say this because I continue to have dead nodes, dropped commands, and laggy interactions if i don't restart the VM after ending a heal early. Regardless of pressing the stop button the network still performs really badly until the VM is bounced.

robertsLando, 2023-01-31:
Fixed in zui 8.8.3

robertsLando, 2023-01-31:
BTW @AlCalzone This could easily be done with a driver function snippet

AlCalzone, 2023-01-31:
@ColColonCleaner Can you capture that problem in a log? I don't have a large network right now, so all I could verify is that the queued commands for battery nodes are discarded.

AlCalzone, 2023-01-31:
I have a PR up that allows skipping the heal for sleeping devices. However, I'd still like to investigate the issue at hand.

ColColonCleaner, 2023-02-02:
Here are log files of healing after the latest update to UI.

1. Restarted the driver so it was fresh
2. waited for all of the nodes to report and commands/messages to settle down
3. started the heal
4. waited for all of the powered nodes to complete
5. stopped the heal
6. noted that the sleeping nodes were marked skipped, nice!
7. waited a couple minutes
8. tried to interact with some of the nodes
9. Several of them lagged so long they got marked dead, and the rest were very slow to operate. It's basically like the heal was still going.

[z-ui_2023-02-01.log](https://github.com/zwave-js/node-zwave-js/files/10564952/z-ui_2023-02-01.log)
[zwavejs_2023-02-01.log](https://github.com/zwave-js/node-zwave-js/files/10564953/zwavejs_2023-02-01.log)

AlCalzone, 2023-02-02:
It seems that all of your issues are somehow entangled. The lag you're seeing is caused by spam from the network while the controller is sending and cannot reply, while at the same time causing the controller to take much longer for sending than normal, here for a response to Node 39:
```
2023-02-02T04:53:20.512Z DRIVER » [Node 039] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x05
                                  │ callback id:      207
                                  └─[SecurityCCNonceReport]
                                      nonce: 0x7f8bddb85bf93676

... some incoming messages ...

... Node 39 repeats its request a bunch of times
2023-02-02T04:53:21.833Z DRIVER « [Node 039] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -72 dBm
                                  └─[SecurityCCNonceGet]
2023-02-02T04:53:21.835Z CNTRLR   [Node 039] in the process of replying to a NonceGet, won't send another NonceR
                                  eport
2023-02-02T04:53:22.451Z SERIAL « 0x010b00a800012702984000b616                                        (13 bytes)
2023-02-02T04:53:22.452Z SERIAL » [ACK]                                                                   (0x06)
2023-02-02T04:53:22.454Z DRIVER « [Node 039] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -74 dBm
                                  └─[SecurityCCNonceGet]
2023-02-02T04:53:22.456Z CNTRLR   [Node 039] in the process of replying to a NonceGet, won't send another NonceR
                                  eport
2023-02-02T04:53:22.536Z SERIAL « 0x010b00a800012702984000b616                                        (13 bytes)
2023-02-02T04:53:22.537Z SERIAL » [ACK]                                                                   (0x06)
2023-02-02T04:53:22.538Z DRIVER « [Node 039] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -74 dBm
                                  └─[SecurityCCNonceGet]
2023-02-02T04:53:22.539Z CNTRLR   [Node 039] in the process of replying to a NonceGet, won't send another NonceR
                                  eport
2023-02-02T04:53:23.186Z SERIAL « 0x010b00a800012702984000aa0a                                        (13 bytes)
2023-02-02T04:53:23.188Z SERIAL » [ACK]                                                                   (0x06)
2023-02-02T04:53:23.189Z DRIVER « [Node 039] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -86 dBm
                                  └─[SecurityCCNonceGet]
2023-02-02T04:53:23.191Z CNTRLR   [Node 039] in the process of replying to a NonceGet, won't send another NonceR
                                  eport
2023-02-02T04:53:23.386Z SERIAL « 0x010b00a800012702984000b717                                        (13 bytes)
2023-02-02T04:53:23.387Z SERIAL » [ACK]                                                                   (0x06)
2023-02-02T04:53:23.388Z DRIVER « [Node 039] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -73 dBm
                                  └─[SecurityCCNonceGet]
2023-02-02T04:53:23.389Z CNTRLR   [Node 039] in the process of replying to a NonceGet, won't send another NonceR
                                  eport
2023-02-02T04:53:23.840Z SERIAL « 0x010b00a800012702984000b717                                        (13 bytes)
2023-02-02T04:53:23.842Z SERIAL » [ACK]                                                                   (0x06)
2023-02-02T04:53:23.843Z DRIVER « [Node 039] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -73 dBm
                                  └─[SecurityCCNonceGet]
2023-02-02T04:53:23.845Z CNTRLR   [Node 039] in the process of replying to a NonceGet, won't send another NonceR
                                  eport
2023-02-02T04:53:23.925Z SERIAL « 0x010b00a800012702984000b717                                        (13 bytes)
2023-02-02T04:53:23.926Z SERIAL » [ACK]                                                                   (0x06)
2023-02-02T04:53:23.927Z DRIVER « [Node 039] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -73 dBm
                                  └─[SecurityCCNonceGet]
2023-02-02T04:53:23.928Z CNTRLR   [Node 039] in the process of replying to a NonceGet, won't send another NonceR
                                  eport
2023-02-02T04:53:24.392Z SERIAL « 0x010b00a800012702984000b616                                        (13 bytes)
2023-02-02T04:53:24.395Z SERIAL » [ACK]                                                                   (0x06)
2023-02-02T04:53:24.397Z DRIVER « [Node 039] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -74 dBm
                                  └─[SecurityCCNonceGet]
2023-02-02T04:53:24.400Z CNTRLR   [Node 039] in the process of replying to a NonceGet, won't send another NonceR
                                  eport
2023-02-02T04:53:27.382Z CNTRLR   [Node 039] Timed out while waiting for a response from the node (ZW0201)

... now 189 wants to be really sure that we get its update:
2023-02-02T04:53:41.504Z DRIVER « [Node 189] [REQ] [BridgeApplicationCommand]
[... 264 lines cut ...]
... more incoming messages ...

... and finally the message reached Node 39 after only 43 seconds:
2023-02-02T04:54:04.569Z DRIVER « [REQ] [SendDataBridge]
[... cut ...]
2023-02-02T04:54:58.392Z DRIVER « [REQ] [SendDataBridge]
                                    callback id:                           215
                                    transmit status:                       Fail, took 0 ms
                                    routing attempts:                      0
                                    protocol & route speed:                Unknown (0x00)
```

I'll reply to your other comment in more detail when I have the time. It surprises me a bit thought that you had a worse experience on 500 series.

AlCalzone, 2023-02-02:
Anyways, the healing process does stop, so I'm going to close this issue in favor of the other one.

closed 2023-02-02 by AlCalzone

bluemantwo, 2023-02-03:
I can confirm healing process stops and no residual traffic is seen.  Able to conduct individual heals immediately, further proving the other heal thread is stopped.  Nice!