zwave-js/zwave-js #5373: Re-turn on  the device after giving the turn off command from UI

Inoy-D, 2023-01-30:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

Hi.
Re-turn on  the device after giving the turn off command from UI

see log.

This does not allow you to use the device

### Device information

Manufacturer: Philio
Model name: PAN03
Node ID in your network: 

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 1.6.0
`node-zwave-js` branch:
`zwave-js-ui` branch:

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[log.log](https://github.com/zwave-js/node-zwave-js/files/10538919/log.log)

Inoy-D, 2023-01-30:
Node id: 048

Inoy-D, 2023-01-30:
yes, this crop from original logfile

AlCalzone, 2023-01-30:
What you see is that the driver confirms the change after attempting it, and the device always returns that it is off.
I can't tell you why that is - maybe you have remote control disabled in the configuration or so?

Inoy-D, 2023-01-31:
[device_debug_info.txt](https://github.com/zwave-js/node-zwave-js/files/10548050/device_debug_info.txt)

more information about device setup

AlCalzone, 2023-01-31:
Ok there is a value "RF protection state", but it seems to be set to "Unprotected".
There's also a config parameter "RF off command mode", which can be set to "Switch on". This sounds a bit like what you're experiencing, but it is supposedly set to "Switch off" already.

You could try changing both to something else and back.
If that doesn't help, I'm afraid this is a case for the manufacturer support.

Inoy-D, 2023-01-31:
The "RF off command mode" parameter determines the device's reaction to the BASIC_SET、BINARY_SWITCH_SET、command.
SWITCH_ALL_OFF.
I checked the behavior of the device with different values of this parameter. The result is the same: when a coma is applied, the device turns off and turns on again after a second. It is necessary to send a shutdown command several times in order for the device to turn off completely. Usually two to 10 times.

I am attaching a new file with debug information of the device and a manual for the device

Thank you for helping me solve the problem

[info2.txt](https://github.com/zwave-js/node-zwave-js/files/10549260/info2.txt)

[PAN03+manual-A1_20210812.pdf](https://github.com/zwave-js/node-zwave-js/files/10549263/PAN03%2Bmanual-A1_20210812.pdf)

AlCalzone, 2023-01-31:
All I can see in the logs is that the device does not seem to switch. Z-Wave JS sends the OFF command, and the device acknowledges that it got the command. Then Z-Wave JS checks the status and the device tells us that it is ON.

I'm unable to help you figure out why this happens. I'm afraid only the manufacturer can do that.

Inoy-D, 2023-02-01:
It is possible to introduce a delay between the command from the state of the state and the request of the state

AlCalzone, 2023-02-01:
There already is a 1 second delay.

Maybe I'm misunderstanding... Does the device actually not turn off or is this just a UI issue?
Because if it the first, adding a further delay won't help. If it's a UI issue because the device is a bit slow, it might.

Inoy-D, 2023-02-01:
Physically, the device manages to turn off but after a couple of seconds it turns on again. Often by giving the command to turn off, you can ensure that the device remains in the off state.

AlCalzone, 2023-02-01:
So this is a device issue, not a driver issue. Delaying the query for the current state won't change this.

closed 2023-02-05 by zwave-js-assistant[bot]