zwave-js/zwave-js #5384: Missing device configuration: DONEXON - varmo TZ eco

adamtyron, 2023-01-31:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

DONEXON - varmo TZ eco

### Manufacturer ID

?

### Product Type

0x0003

### Product ID

0x0004

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/3328?selectedFrequencyId=-1

AlCalzone, 2023-01-31:
Duplicate of #5204

AlCalzone, 2023-01-31:
FWIW, this device was added in v10.5.2

closed 2023-01-31 by AlCalzone