zwave-js/zwave-js #5398: HEATIT Z-SMOKE MULTISENSOR - Unknown product 0x0017

dffffffff, 2023-02-01:
zwave-js-ui: 8.8.2.b5fcc4a
zwave-js: 10.5.4

HEATIT Z-SMOKE MULTISENSOR when added will return 
Heatit Controls AB - Unknown product 0x0017 (0x0003)

[image]

Fingerprint:
manufacturer: Heatit Controls AB (0x019b)
product type: 0x03
product id:   0x17

driver log: http://paste.debian.net/hidden/f5b84956/
zwave log: http://paste.debian.net/hidden/39a5f4f3/

AlCalzone, 2023-02-01:
Is this the [battery powered](https://devices.zwave-js.io/?jumpTo=0x019b:0x0003:0x3602:0.0) or the [mains powered](https://devices.zwave-js.io/?jumpTo=0x019b:0x0003:0x3601:0.0) version?

kailampi, 2023-02-01:
I would say it is AC powered. I have very same issue with AC powered  version.

[image]

AlCalzone, 2023-02-01:
@zwave-js-bot add fingerprint `0x0003:0x0017` to `0x019b:0x0003:0x3601:0.0`

PR #5401 (2023-02-01): fix(config): add fingerprint `0x0003:0x0017` to "Heatit Z-Smoke 230V"
fixes: #5398
closed 2023-02-01 by AlCalzone