zwave-js/zwave-js #5399: Aeon WallMote Quad ZW130 Quad Mote Swipe

Sian-Lee-SA, 2023-02-01:
I did a quick search regarding this device but there was no issue relating to the swipe function of the device. Previosuly OpenZWave was able to intercept this action but a custom script was needed to make better use of the function.

I can not find any event related to the swipe down or swipe up action for this device with Z-Wave JS. The config also has this option enabled.

Sian-Lee-SA, 2023-02-02:
Thanks, That works except theres just one problem, I can't differentiate if the swipe was from the left or right side.

AlCalzone, 2023-02-02:
You're right. We don't pass the endpoint information with those notifications.

When we do, you need to set up a multi channel association for the buttons:
[image]

Sian-Lee-SA, 2023-02-02:
Ok, at first there was nothing in the endpoint 1-4. So I figured that I would reinterview the node (because why not?). All of a sudden I had 4 entries added automatically in the group for the Dimmer key (endpoints 1-4). I then did a test and I was getting duplicate notifications. So I removed those multileveled buttons and tested.

I can see the event like from the button (multilevel) was doing but nothing through Home Assistant anymore unfortunately. Sorry, I'm more experienced with Zigbee than Z-Wave and these things are similar but not everything will be clear right away so excuse my misunderstandings.

[image]

[image]

closed 2023-09-06 by AlCalzone