zwave-js/zwave-js #5413: zwave-js is unable to update user codes on Schlage Connect™ Smart Deadbolt

parkervcp, 2023-02-05:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
applying a usercode change

What do you observe?
When attempting to update a user code it fails and ZWave-JS-UI shows an error showing `Driver: User Code: "userCode" must be of type "string or Buffer", received "number" (ZW0322)`

The service logs from zwave-js-ui - https://gist.githubusercontent.com/parkervcp/17787f1f965e092185f2c302ebdd9c0b/raw/8bd6f5c45be821af2cbcd77039ef5c0577541470/zwavejs%2520silly%2520logs

debug logs from home assistant - https://gist.githubusercontent.com/parkervcp/e289ee1a4dc4ffa934fbc72c395af82e/raw/745ca099ceef76da23fc9a115424b78ac08d67da/gistfile1.txt

What did you expect to happen?
userCodes updating to the new value

Steps to reproduce the behavior:
  1. Go to Control Panel
  2. Click on Touchscreen Deadbolt Z-Wave Plus 
  3. Go to User Code
  4. Change User Code for one user
  5. Send New Code to Deadbolt
  6. Get error `Driver: User Code: "userCode" must be of type "string or Buffer", received "number" (ZW0322) ` in zwave-js-ui

### Device information

Manufacturer: Allegion
Model name: BE469ZP 
Node ID in your network: 3

AKA - Schlage Connect™ Smart Deadbolt

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwave-js-ui: 8.8.3
zwave-js: 10.5.5
home id: 3512255105
home hex: 0xd158c281

### Did you change anything?

no

### If yes, what did you change?

No changes made

### Did this work before?

Yes (please describe)

### If yes, where did it work?

I haven't had to change user codes in at least 6 months so this is a new issue from then to now

### Attach Driver Logfile

[zwavejs_2023-02-04.log](https://github.com/zwave-js/node-zwave-js/files/10610174/zwavejs_2023-02-04.log)

AlCalzone, 2023-02-06:
How exactly are you changing user codes? I tried it myself (different lock but should not matter) and could not reproduce.

parkervcp, 2023-02-07:
Using both the zwave-js-ui and via the developer service page fails to update user codes. I also tried a re-interview per the troubleshooting section and now I don't even have the ability to change usercodes.

Here is a screenshot in the zwave-js-ui
[image]

AlCalzone, 2023-02-07:
Try going into settings and check the box "increase report timeouts". This lock sometimes takes ages to respond all of a sudden. Then re-interview with the box "reset security classes" checked.

parkervcp, 2023-02-08:
Well that would seem to be what fixed the issue entirely @AlCalzone

closed 2023-02-08 by parkervcp