zwave-js/zwave-js #5428: FGS214 - Endpoint not visible in ZwaveJS even though it is well declared in config file

anto35, 2023-02-07:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

I need to associate a FGS-214 with another module in the "On/ Off S2" group. This group is normally attached to endpoint 2 (indicated in the FGS-214 user manual).

In Zwave-JS UI, endpoint 0 is the only one present. Endpoints 1 & 2 are ignored. 

They are however declared in the zwave-js config file of the FGS-214.

### Device information

Manufacturer: 0x10f
Model name: FGS-214

Please find attached in the driver log field the debug info of the node

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwave-js-ui: 8.8.3
zwave-js: 10.5.5

### Did you change anything?

yes (please describe)

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

 {
  "id": 100,
  "name": "Door Switch",
  "loc": "Garage",
  ****"groups": [
    {
      "text": "Lifeline",
      "endpoint": 0,
      "value": 1,
      "maxNodes": 1,
      "isLifeline": true,
      "multiChannel": true
    },**
    {
      "text": "On/Off S1",
      "endpoint": 0,
      "value": 2,
      "maxNodes": 5,
      "isLifeline": false,
      "multiChannel": true
    }**
  ],
  "neighbors": [],
  "ready": true,
  "available": true,
  "hassDevices": {},
  "failed": false,
  "inited": true,
  "status": "Alive",
  "interviewStage": "Complete",
  "hexId": "0x010f-0x0404-0x1000",
  "dbLink": "https://devices.zwave-js.io/?jumpTo=0x010f:0x0404:0x1000:5.0",
  "manufacturerId": 271,
  "productId": 4096,
  "productType": 1028,
  "deviceConfig": {
    "filename": "/data/db/devices/0x010f/fgs214.json",
    "isEmbedded": true,
    "manufacturer": "Fibargroup",
    "manufacturerId": 271,
    "label": "FGS214",
    "description": "Smart Module",
    "devices": [
      {
        "productType": 1028,
        "productId": 4096
      },
      {
        "productType": 1028,
        "productId": 12288
      },
      {
        "productType": 1028,
        "productId": 16384
      }
    ],
    "firmwareVersion": {
      "min": "0.0",
      "max": "255.255"
    },
    "associations": {},
    "paramInformation": {
      "_map": {}
    }
  },
  "productLabel": "FGS214",
  "productDescription": "Smart Module",
  "manufacturer": "Fibargroup",
  "firmwareVersion": "5.0",
  "protocolVersion": 3,
  "zwavePlusVersion": 1,
  "zwavePlusNodeType": 0,
  "zwavePlusRoleType": 5,
  "nodeType": 1,
  "endpointsCount": 0,
  "endpointIndizes": [],
  "isSecure": true,
  "security": "S2_Authenticated",
  "supportsSecurity": false,
  "supportsBeaming": true,
  "isControllerNode": false,
  "isListening": true,
  "isFrequentListening": false,
  "isRouting": true,
  "keepAwake": false,
  "maxDataRate": 100000,
  "deviceClass": {
    "basic": 4,
    "generic": 16,
    "specific": 1
  },
  "firmwareCapabilities": {
    "firmwareUpgradable": true,
    "firmwareTargets": [
      0
    ],
    "continuesToFunction": "unknown",
    "supportsActivation": "unknown"
  },
  "deviceId": "271-4096-1028",
  "statistics": {
    "commandsTX": 6,
    "commandsRX": 5,
    "commandsDroppedRX": 0,
    "commandsDroppedTX": 0,
    "timeoutResponse": 0,
    "rtt": 46.5
  },
  "lastActive": 1675775873300,
  "_name": "Switch Porte (Garage)"
}

anto35, 2023-02-07:
[node_100.json.txt](https://github.com/zwave-js/node-zwave-js/files/10677222/node_100.json.txt)

AlCalzone, 2023-02-07:
Please read the bot's link again. What you posted is not a driver log. While you're making the logfile, re-interview the device and wait for that to finish.

anto35, 2023-02-07:
PLease find attached the requested logs : 

[zwavejs_2023-02-07.log](https://github.com/zwave-js/node-zwave-js/files/10679540/zwavejs_2023-02-07.log)

AlCalzone, 2023-02-07:
The device does not have multiple endpoints. Is it possible you missed this in the manual?

[image]

In any case you can just ignore the bits about endpoints and define your associations on Endpoint 0 (the Root):
[image]

I'm afraid you're out of luck for the Switch S2 part, that seems to be limited to the Double Switch:
[image]

anto35, 2023-02-08:
Hello and thank you for your answer.

I had indeed missed the information. If I understand correctly, on the FGS-214, the S2 switch can only be used to launch scenes.

I'll do the test with a FGS-224 tonight and I'll complete this ticket because it seems to me that I didn't have the multiple endpoints on the FGS-224 either.

closed 2023-02-11 by zwave-js-assistant[bot]