zwave-js/zwave-js #5431: Support for rfc2217

robkamp, 2023-02-08:
It wold be nice to allow for and USB device like a z-wave dongle to be mounted over ip. This can be done by implementing the  rfc2217 feature set.

AlCalzone, 2023-02-08:
Haven't read the RFC but you can point the driver at a hosted serialport, e.g. with ser2net. Just use a port setting like this:
```
tcp://hostname:port
```

robkamp, 2023-02-08:
Works as described using tcp 

my YAML for ser2net is as follows:

`define: &banner \r\nser2net port \p device \d [\B] (Debian GNU/Linux)\r\n\r\n

connection: &racwon115200
    accepter: tcp,3334
    enable: on
    options:
      banner: *banner
      kickolduser: true
      telnet-brk-on-sync: true
    connector: serialdev,
               /dev/ttyAMA0,
               115200n81,local`

closed 2023-02-08 by robkamp