zwave-js/zwave-js #5433: Add group

BrukerJan, 2023-02-08:
I'm not able to add groups to a node. Adding seems to go through and messages (green, in upper right corner) tells that everything went well. Am I doing it wrong, or can it be a bug?
[image]

AlCalzone, 2023-02-08:
Again a device that doesn't support Supervision properly 🙈

BrukerJan, 2023-02-08:
Could you please be a little more specific about this? I am a former IT professional, but have little experience with z-wave and don't quite see how to proceed to solve my problems.

AlCalzone, 2023-02-08:
You can't. I need to disable supervision for that device in a config file.

AlCalzone, 2023-02-08:
@zwave-js-bot add compat flag to `0x019b:0x0003:0x0208:0.0`
```jsonc
"commandClasses": {
	"remove": {
    	// The device does not support Supervision correctly
        "0x6c": {
			"endpoints": "*"
		},
	}
}
```

PR #5435 (2023-02-08): fix(config): disable Supervision for Heatit Z-RELAY
fixes: #5433

**TODO:**
- [x]	Change PR title to be more specific
closed 2023-02-08 by AlCalzone