zwave-js/zwave-js #5445: Ultraloq Pro Z-wave

JPDMaker, 2023-02-09:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

[77-98-0-latchStatus] Current status of the latch is always reporting "OPEN"

U-tec support said the issue could be on the z wave hub side.
Thread of others reporting the same issue: https://community.u-tec.com/t/expand-and-fix-z-wave-capabilities/502/18

### Device information

Manufacturer: u-tec
Model name: U Bolt Pro Z wave 
Node ID in your network: <077>

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwave-js-ui: 8.6.3.c12b1c6
zwave-js: 10.3.1

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwavejs_current.log](https://github.com/zwave-js/node-zwave-js/files/10701445/zwavejs_current.log)

efaden, 2023-02-11:
Having same issue

efaden, 2023-02-11:
[image]

AlCalzone, 2023-02-11:
I'll try on Monday

efaden, 2023-02-11:
Thanks.  There really should be two entities for the pro if the sensor is present.  One for the lock, and one for the door sensor

AlCalzone, 2023-02-13:
That's a bug in the device firmware, at least in v1.1.3 which my lock is on.
When being queried for the supported door components, it has the byte set to 6:
[image]

which means Bolt and Latch states are supported, Door state isn't:
[image]

Given that the manual states that Bolt and Door are supported, this seems like a bug.

AlCalzone, 2023-02-13:
Despite that, I can't get the lock to report any change in door state.

efaden, 2023-02-13:
Well that's not awesome.  I send an email to them about this issue with this thread... Maybe they can fix it.  Would be great to be able to get door status

tekkenly, 2023-03-06:
Hi @AlCalzone ,

I wonder if we can put some special code in to make a fix for this device and this version firmware? the outdoorhandle value is clearly matching the door sensor state.

Or, if you can give me a quick hint where the related code is, I can try to make a change, then you can choose if you want to merge it in.

Thanks.

AlCalzone, 2023-03-06:
We don't hardcode such one-time device-specific behavior.

AlCalzone, 2023-05-15:
I've heard that U-Tec finally managed to reproduce the issue, but no idea if and when this will be fixed.

johntdyer, 2023-06-05:
@AlCalzone  do you have a link to their issue I can follow ?

AlCalzone, 2023-06-06:
Nope, that was all back-channel information. Haven't head anything since.

TheOneOgre, 2023-07-04:
I would like to add that they have seemingly made no progress on fixing this issue and I have discovered a second issue and it seems others have as well.

When the lock is triggered to lock, even if the locking process fails, say the door isn't closed all the way or something happens to make it not engage the deadbolt, the lock will beep and error out but the reported status to z-wave is that the lock was successful. This is dangerous as even though you should always manually check for security, it should report some kind of error or at least fallback to the previous state. Kind of disappointed in Ultraloq on the development of this.

efaden, 2023-07-04:
Yeah, the lock is good, but if they don't actually fix bugs then it's
really not a great choice

balloob, 2023-07-05:
We have decided to terminate the Works with Home Assistant relationship with Ultraloq. We had reported these issues with them and were discussing solutions. However, they ceased communication from their end and have received no further updates.

TheOneOgre, 2023-07-05:
> We have decided to terminate the Works with Home Assistant relationship with Ultraloq. We had reported these issues with them and were discussing solutions. However, they ceased communication from their end and have received no further updates.

Good to see some action being taken about this. Thanks for keeping Home Assistant great!

AlCalzone, 2023-07-05:
I'm afraid there's nothing to be done in this project, since the functionality is just broken.

closed 2023-07-05 by AlCalzone

holblin, 2023-07-07:
Could we update https://www.home-assistant.io/integrations/ultraloq/ ?

b-uwe, 2023-07-09:
I'll do that on Tuesday!

b-uwe, 2023-07-11:
@TheOneOgre @holblin @AlCalzone https://github.com/home-assistant/home-assistant.io/pull/28166

hongjun1986, 2023-08-15:
This problem has been solved in the new firmware, which can be upgraded online through Home assistant.

God-TM, 2023-08-23:
Will it be added back into the WWHA partner program?

hongjun1986, 2023-08-23:
Yes, we do our best to add it back.

Thanks!

***@***.***
 
From: God-TM
Date: 2023-08-23 08:51
To: zwave-js/node-zwave-js
CC: jack; Comment
Subject: Re: [zwave-js/node-zwave-js] Ultraloq Pro Z-wave (Issue #5445)
Will it be added back into the WWHA partner program?

TheOneOgre, 2023-08-23:
> Will it be added back into the WWHA partner program?

I wouldn't think it would or should be added back. Ignoring HA devs, not providing timely updates, not notifying users of issues, and overall not being easy to work with, is an easy disqualifier for being in a partner program. I think the biggest thing was the untimely manner in which they handled this and the complete lack of communication.

Also, I haven't tested it yet but, can anyone confirm if the "false lock" bug is still occurring where if the lock is told to lock, but fails due to the door being slightly open or something, the state is still reported as "Locked"? Was a HUGE oversight before if you relied on that status for your door being locked.

brettmiller, 2023-08-23:
> Also, I haven't tested it yet but, can anyone confirm if the "false lock" bug is still occurring where if the lock is told to lock, but fails due to the door being slightly open or something, the [...]

Tested and the "false lock" bug still exists.

b-uwe, 2023-08-23:
> > Will it be added back into the WWHA partner program?

> I wouldn't think it would or should be added back. Ignoring HA devs, not providing timely updates, not notifying users of issues, and overall not being easy to work with, is an easy disqualifier for being in a partner program.

Things have changed drastically in the last few weeks! Please feel free to put that to a test! I' doing exactly so with the last statement, the "false lock" bug which I expect to be worked on very soon.
But that's why, yes, we plan to add them back to the WWHA partner program!

hongjun1986, 2023-08-23:
> Tested and the "false lock" bug still exists.

Thanks for Testing.  I will check it ASAP.

God-TM, 2023-08-24:
I just tested it. Perhaps I'm not understanding what's wrong here, but when I open and close the door, the sensor labled "Current status of the door" goes from Open to Closed and vice-versa accurately reflecting the state of the door. I can tell it to lock in either state, and the state is still either Open or Closed depending on if it's actually open or closed...

hongjun1986, 2023-08-24:
I have tested it, the door status is correct depending the open or close operation.  
Please put the door magnet as close as possible to the lock, the farther the magnet is, the weaker the magnetic is.
Can you show me your test environment picture  include the lock and door magnet ？

God-TM, 2023-08-24:
I used to have a Gen 2 August Pro there, so I left the magnet from that and used it with the Ultraloq Pro. So far it's been working without any issues.
[image]
[image]
[image]
[image]
[image]

Please let me know if I can take any better measurements or angles that might be helpful for you.

hongjun1986, 2023-08-24:
Thanks for your info.
1.  Can you gvie your lock Serial Number （SN：xxxxx）,inside of the battery cover ?
2. The magnet detection chip in the lock  and the detection algorithm are different from August, we don't  do compatibility test with August magnet.  Can you install ultraloq magnet as installation guide and test again？

God-TM, 2023-08-24:
I'm not the one having issues with this lock... I'd rather not break what's currently working for me.

JPDMaker, 2023-08-24:
I tested the lock jam bug. Does not update status. 

lock serial number: 8030305122130050

[image]
[image]

hongjun1986, 2023-08-25:
Do our best to optimizing them.

hongjun1986, 2023-09-04:
hi JPDMaker，
   Where should the status of Jam be displayed  in home assistance UI?

JPDMaker, 2023-09-04:
It’s the state of the lock entity. Should update state to either ; Jammed, Locked, Locking, Unlocked, Unlocking, Unavailable, Unknown

[https://developers.home-assistant.io/docs/core/entity/lock/](https://developers.home-assistant.io/docs/core/entity/lock/)

HarlemSquirrel, 2023-09-05:
Oh we definitely should have those states!

HarlemSquirrel, 2023-09-05:
Took me a while to track this down but this appears to be set in https://github.com/home-assistant-libs/zwave-js-server-python/blob/0.51.1/zwave_js_server/const/command_class/lock.py#L81-L86

Then mapped in HA core in https://github.com/home-assistant/core/blob/2023.8.4/homeassistant/components/zwave_js/lock.py#L40-L49

AlCalzone, 2023-09-05:
Jammed is not a lock state in `node-zwave-js`. It's a notification variable, which gets translated to a sensor in HA, I think here:
https://github.com/home-assistant/core/blob/7548c4aced3872c0d685ae0b07529a1f20fddbcd/homeassistant/components/zwave_js/discovery.py#L773-L785

hongjun1986, 2023-09-12:
This issue( When the lock is jammed,the lock status is lock, user can not lock again) is fixed. We are trying to add the jammed notifacation funcation, need more time.

itsensellc, 2023-10-13:
New Z-Wave U-Bolt Pro, latest firmware applied, status is still stuck at open in HA. The app shows correctly so the position sensor is working.

AlCalzone, 2023-10-16:
I still find it tricky to trigger changes of the sensor without mounting the lock in a doorframe, but I haven't managed to get the state in the app and HA out of sync. When the status does change, I see it in both. Might be a connectivity issue?

itsensellc, 2023-10-16:
> I still find it tricky to trigger changes of the sensor without mounting the lock in a doorframe, but I haven't managed to get the state in the app and HA out of sync. When the status does change, I see it in both. Might be a connectivity issue?

Don't think so? After latest update the door status seems to be working slightly better. What consistently seems to not work is auto-lock after X seconds doesn't seem to update the lock status.  Opening the door from outside and then it locking automatically when closed is working. But if I open it from HA, then it auto locks after 15 seconds... the status does not get updated.

AlCalzone, 2023-10-16:
> auto-lock after X seconds doesn't seem to update the lock status

Wait, are we talking about the lock status or the door status?

itsensellc, 2023-10-16:
> > auto-lock after X seconds doesn't seem to update the lock status
> 
> Wait, are we talking about the lock status or the door status?

Sorry I kind of switched gears. Door open status seems to be a bit better since the latest update. However the auto-lock has been an issue since adding the lock. But anytime I manually unlock it, that gets tracked properly in HA.

AlCalzone, 2023-10-16:
Ok I'll try to reproduce that tomorrow

hongjun1986, 2023-10-17:
I have tested in our site. The lock updates the correct status after auto-lock.   For testing door status,please move the lock body,don't move magnet.

The version of my lock.
The firmware version(in u-tec app show):02.26.0022
The Z-wave firmware version(HA show):1.5.3

Can you share me the testing video? We want more operate details to debug the issue. Thanks.

[image]

itsensellc, 2023-10-17:
> I have tested in our site. The lock updates the correct status after auto-lock. For testing door status,please move the lock body,don't move magnet.
>
> The version of my lock. The firmware version(in [...]

Ok I have some further testing results. Door open status seems to be working perfectly fine for the moment. Door lock status status when opened at the lock seems perfectly fine, including when it auto locks. When I open the lock via the app, and it auto locks, status is reported fine. It's only when I open the door via HA that this issue occurs. And it doesn't seem to matter which kind of card I use. I can open the lock, it will update the status, but when the lock auto locks the HA state still reflects unlocked. When I click it again, it will beep and indicate lock. Sometimes this can get totally out of sync where it think it's unlocked but it's actually locked or vice versa. So it's only when controlling via HA that this is occurring. I can try and make a video if it's helpful I just wanted to clarify that first.

hongjun1986, 2023-10-17:
Can you give me a interface picture that you operate the lock and the status show ? 
I will check it.

itsensellc, 2023-10-17:
> Can you give me a interface picture that you operate the lock and the status show ? I will check it.

I will try and upload something a bit later. I did want to add I found something else interesting today. I opened the lock from inside to go out. It captured the door and lock sensors fine. However, it did not capture either door unlock/lock or open/closed events when I came back, using the fingerprint sensor and the door automatically locking when I shut it. I'm not sure if maybe it just didn't have time to register those events or they didn't get sent or what.

hongjun1986, 2023-10-17:
My testing enviroment is:  a  Iaptop ( installed  HA virtual machine, Connected a SLUSB700A  hub with new driver, edge Brower  
 to login to HA in virtual machine and do operation  ); a u-bolt pro z-wave and magnet on a door. I don't use mobile APP.
what's your testing enviroment?  Where do you want to see the events? Let's keep in the same page.

itsensellc, 2023-10-18:
> My testing enviroment is: a Iaptop ( installed HA virtual machine, Connected a SLUSB700A hub with new driver, edge Brower to login to HA in virtual machine and do operation ); a u-bolt pro z-wave [...]

Ya not trying to muddy the waters. Just let me know what helps most to troubleshoot.

I'm running a dedicated ASRock BeeBox, with HAOS running Virtualbox (latest version, extension manager installed and USB passthrough), on Windows 10, Zooz ZST10-700 with firmware 7.19.2. Doesn't matter if I login with the HA web interface, or the HA app., I get the same result. I have the magnet on the door frame as well.

I am looking at the entity values/history directly for the most part. I know if those don't work then anything relying on them most likely won't either.

hongjun1986, 2023-10-18:
Sorry, I just want to know how to reproduce the bug you mentioned in my side and try  help debug. I will update If i have new info.

itsensellc, 2023-10-18:
100%! No need to apologize I appreciate your help. Will get you info soon.

creeve4, 2023-11-27:
Any update on the last remaining bug?

itsensellc, 2023-12-01:
> Any update on the last remaining bug?

Hi Chris - apologies I was travelling for the better part of the last month and a half. I will shoot a video for you and post it. It is still happening for sure.

itsensellc, 2023-12-01:
Here's a quick video to illustrate what is happening: [https://youtu.be/cmq1fcqTxyA](url)

In short: if I click the unlock button, the lock unlocks as expected. Sometimes it takes a second to register on the card in HA but that works pretty consistently. But notice once the door lock auto locks (10 seconds after the unlock), the status doesn't update in HA. It remains as "unlocked". If I click "lock" again in HA you'll notice you hear one beep and the status updates in HA. I guess the lock knows it's locked hence the single beep. For whatever reason HA just won't update with that manual action. This only happens when I use the unlock button on one of the cards. If I use the Utec app, the status updates for both when it unlocks and auto locks as expected.

holblin, 2023-12-01:
I agree, I have the same behavior in my instance.

creeve4, 2023-12-02:
Thank you for sharing the video, I am experiencing the same with my lock.

zirdum, 2023-12-02:
Interesting. Mine isn't doing that.

I click the unlock button, it unlocks. Mine is set to auto lock after 30 seconds. It locks and the status in HA updates immediately.

Interesting. If I set it to lock after 10 seconds, it does what you're experiencing. Setting it to 20 seconds works fine.

Hopefully this helps with troubleshooting the bug.

[10 more replies]