zwave-js/zwave-js #5450: Network health checks don't include bidirectional pings most of the time

ColColonCleaner, 2023-02-10:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

8.8.0

### ZwaveJS version

10.4.0

### Describe the bug

Most of the time, node health checks directed at the controller only include pings from the controller to the nodes and not the other way around.

Rarely I'll get a health check that does include pings from the node to the controller and that communication is universally terrible in comparison to the traffic from controller to node.

This isn't related to specific devices that may or may not support bidirectional pings because ill run a health check on the same device many times and only once in a dozen or so will it attempt bidirectional pings.

And then there are instances where it mixes checks of one way pings with bidirectional pings and that results in wild fluctuations of ratings, like this screenshot:

### To Reproduce

Not sure how this could be reproduced. Should I always be expecting attempts of bidirectional pings?

### Expected behavior

Bidirectional pings are always attempted since the node comms are equally important to controller comms.

Hooverjc, 2023-02-10:
As @robertsLando linked to this, similar issue for me it appears. since moving back to the 500 stick as indicated above, I have seen much less issues, but was doing some health checks and noticed some of the same behavior with devices nearest to the controller and I wonder if this isn't causing the issue with my devices at 3-4 hops reporting "Failed to send the command after 3 attempts (Status NoAck) (ZW0204)" They work most of the time and when not marked as dead, work just fine with near 0 latency, but they often just randomly get marked as dead with the above error and come back to life with a ping. looking at the devices in the middle between the controller and these edge devices, several of them have this issue where the controller can talk to them, but they cannot reliably talk back to the controller.

ColColonCleaner, 2023-05-20:
@AlCalzone any news on implementation of application routes?

Hooverjc, 2023-06-02:
> @AlCalzone any news on implementation of application routes?

I see priority routes were added so you can define them. In some of my problematic nodes, the controller decided going to devices on opposite ends of the house was "ideal" and no amount of healing could convince it otherwise. Telling the device to go directly to the controller or through an alternate route has generally improve my health checks and responsiveness. I'd suggest checking this out and see if it helps you at all. Can be found in the UI under the network graph, click the node you want to set a priority route for and then click the set button order your routes and test away. They apply instantly so adding and deleting with a health check in betwene is quick and easy. 
[image]