zwave-js/zwave-js #5456: Zooz ZSE18 Motion Sensor Frequent Dropped Messages

jshtz4, 2023-02-11:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Motion (and no motion/cleared) events from the Zooz ZSE18

What do you observe?
Frequently (10% of the time) some or all parts of the motion/clear event fail to report. The log typically mentions "Dropping message with invalid payload" and/or "error: Duplicate command"

What did you expect to happen?
Motion/clear events to be reported consistently.

Steps to reproduce the behavior:
Trigger motion from a ZSE18, wait for motion to clear. Repeat until failure is observed.

### Device information

Manufacturer:  Zooz
Model name:  ZSE18
Node ID in your network:  009

I have two of these devices and both exhibit this behavior. Only Node 009 is represented in the example logs.

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwave-js-ui: 8.8.5.2d9d59c
zwave-js: 10.10.0
home id: 3780608751
home hex: 0xe15782ef

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[examples.log](https://github.com/zwave-js/node-zwave-js/files/10714662/examples.log)

AlCalzone, 2023-02-11:
> error: Duplicate command

means the device is using an S2 sequence number it recently used (last 10 or so). Since this appears to happen after it requests a nonce (got out of sync somehow), I suspect it uses the sequence number of the nonce request for the following report. Unfortunately, I need to add some logging to know why the message is detected as duplicate.

I'd probably need to see the entire logfile (all communication between the examples).

jshtz4, 2023-02-11:
[zwavejs_2023-02-11.log](https://github.com/zwave-js/node-zwave-js/files/10714834/zwavejs_2023-02-11.log)

Here's a complete log section covering about 20 minutes with 11 instances of "error: Duplicate command"

AlCalzone, 2023-02-13:
If you really didn't cut any communication with Node 9 out, this smells a lot like a device bug.

The sensor requests a new nonce for every single report, which is not necessary for S2, because the shared encryption state is reused for further communication. Also, the sensor seems to randomly skip sequence numbers, almost as if it was restarting after every report and resetting its internal state:

99, 100, 101:
```
2023-02-11T19:49:33.894Z DRIVER « [Node 009] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -74 dBm
                                  └─[Security2CCNonceGet]
                                      sequence number: 99
...
2023-02-11T19:49:34.128Z DRIVER « [Node 009] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -78 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 100
                                    │ extensions:
                                    │ · type: SPAN
                                    │   sender EI: 0x6ead26f89bfd0a2021209164db75e7d3
                                    └─[NotificationCCReport]
                                        notification type:   Home Security
                                        notification status: 255
                                        notification state:  Motion detection
...
2023-02-11T19:49:34.671Z DRIVER « [Node 009] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -79 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 101
                                    └─[BinarySensorCCReport]
                                        type:  Motion
                                        value: true
```

and the next reports just 10s later uses 22, 23, 24:
```
2023-02-11T19:49:44.640Z DRIVER « [Node 009] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -75 dBm
                                  └─[Security2CCNonceGet]
                                      sequence number: 22
...
2023-02-11T19:49:45.008Z DRIVER « [Node 009] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -75 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 23
                                    │ extensions:
                                    │ · type: SPAN
                                    │   sender EI: 0xd20fc0df940eb83da1bc05eba0a30a9f
                                    └─[NotificationCCReport]
                                        notification type:   7
                                        notification status: 255
                                        notification state:  idle
                                        event parameters:    0x08
...
2023-02-11T19:49:45.559Z DRIVER « [Node 009] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -77 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 24
                                    └─[BinarySensorCCReport]
                                        type:  Motion
                                        value: false
```

This might be a shot into the blue, but have you tried changing batteries?

jshtz4, 2023-02-13:
I have two of these devices that both exhibit this behavior when on battery power and included with S2. The ZSE18 supports power via USB as well so I moved one over to USB and excluded/included it as a wired device. On USB power, the sequence numbers are consistent, I haven't seen any duplicate command errors yet, and after the inclusion process there haven't been any additional nonce requests.

I take this all to mean that the problem has to do specifically with how the device communicates using S2 as a battery powered device. Are there any other tests I can perform or logs I can provide?

webmogul1, 2023-02-13:
@jshtz4 What version is the firmware of the ZSE18?

jshtz4, 2023-02-13:
I'm on firmware v1.03.

AlCalzone, 2023-02-14:
> I take this all to mean that the problem has to do specifically with how the device communicates using S2 as a battery powered device.

My assumption would be that the device frequently restarts (crashes?) on battery and just selects a random sequence ID on startup.

Don't think there's much you can test. I'll reach out to my contact at Zooz.

jshtz4, 2023-03-03:
Zooz let me know that they are working on a firmware update to address this issue with the device.

closed 2023-03-07 by zwave-js-assistant[bot]

InstigatorX, 2023-07-13:
@jshtz4 Any word from Zooz? I'm experiencing the same issue.

jshtz4, 2023-07-15:
Unfortunately, no. I was able to run USB power to mine and they've both been perfect since then. I know that's not ideal and hope that Zooz will eventually release the promised firmware update so that the device works as advertised. I'd suggest opening a ticket with them to increase attention on Home Assistant users and also they should notify you directly via that ticket when a firmware update does become available.

InstigatorX, 2023-07-15:
Thx. I opened a ticket a couple days ago and they said it was in the works.