zwave-js/zwave-js #5463: Interview of AEON Labs ZW062 fails with subsystemType is not a SubsystemType

ccie1441, 2023-02-14:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
AEON Labs ZW062 device sits at "NodeInfo". Selecting the node and advanced -> interview shows an error in the logs.

What do you observe?
Node is interviewed up to this point:
2023-02-14T02:00:06.015Z CNTRLR » [Node 024] Querying signaling capabilities...
2023-02-14T02:00:06.070Z CNTRLR « [Node 024] Received supported subsystem types:
· Visual
**· unknown (0x03)**
2023-02-14T02:00:06.073Z CNTRLR » [Node 024] Querying event signaling state for subsystem Visual...
2023-02-14 15:00:06.133 INFO Z-WAVE: [Node 024] Metadata updated: 102-0-signalingState-2
2023-02-14 15:00:06.134 INFO Z-WAVE: [Node 024] Value added: 102-0-signalingState-2 => 255
2023-02-14T02:00:06.139Z CNTRLR « [Node 024] Subsystem Visual has state On
2023-02-14T02:00:06.140Z CNTRLR » [Node 024] Querying event signaling state for subsystem unknown (0x03)...
2023-02-14T02:00:06.141Z CNTRLR [Node 024] **Error during node interview: subsystemType is not a SubsystemType (
ZW0322)**

What did you expect to happen?
Expected the interview to complete and the device to be available for use as it did in previous versions.

Steps to reproduce the behavior:
  1. Go to the ZW062 node in the control panel
  
  3. Click on it and advanced
  4. Click on re-interview
  5. Node sits at NodeInfo and I see the error in the logs

### Device information

Manufacturer: AEON Labs
Model name: ZW062 (Aeon Labs Garage Door Controller Gen5)
Node ID in your network: 024

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwave-js-ui: 8.8.3
zwave-js: 10.5.5
home id: 4287188722
home hex: 0xff894ef2

### Did you change anything?

yes (please describe)

### If yes, what did you change?

Was working with the previous z-wave JS integration and HomeAssistant

### Did this work before?

No, it never worked anywhere

### If yes, where did it work?

Worked with the previous z-wave JS integration, but failed with the z-wave JS UI

### Attach Driver Logfile

[zwavejs_2023-02-14.log](https://github.com/zwave-js/node-zwave-js/files/10728139/zwavejs_2023-02-14.log)

AlCalzone, 2023-02-14:
Ok the root of this issue is that the device incorrectly encodes the support bitmask:
[image]

The specs only know the subsystem types 1 and 2, which means the bitmask should be a single byte with the lowest 2 bits set: `0b0000_0011`. The device however responds with `0b0000_0110`, which includes an invalid value.

The problem you're seeing is because our API has a check for valid subsystem types, but the supported types returned by the device are passed in there without checking first.

closed 2023-02-14 by AlCalzone

ccie1441, 2023-02-14:
Thank you for looking at this so quickly!

AlCalzone, 2023-02-14:
No clue, maybe the implementation does something different when S0 is involved. But yeah, you're right - a barrier operator should be included securely.

ccie1441, 2023-02-14:
The FW reported for the ZW062 is FW: v1.11 (from the Z-Wave JS UI console). Is that the same as the working one?
When setting the node up I have just included, not specifically trying to be secure (or not). I can try that if that is a difference between the node responding correctly or not.

ccie1441, 2023-02-16:
Thanks for looking into this. An upgrade to FW v1.14 on the door controller also has fixed my issue.

PR #5465 (2023-02-14): fix: skip queries for invalid enum members during interview
fixes: #5463