zwave-js/zwave-js #5464: Missing device configuration: Eurotronic Comet Z-Wave

cadwizzard, 2023-02-14:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Eurotronic Comet Z-wave

### Manufacturer ID

0x0148

### Product Type

0x0004

### Product ID

0x0003

### Firmware Version

11.26.4

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/1601

### Do you have a manual?

https://manual.zwave.eu/backend/make.php?lang=en&sku=EURECOMET&cert=

*The device is detected but manufacturer does not show in Home Assistant, looks like a product ID increment as the config DB has product 1,2 and not 3*
The parameters are enumerated in discovery, as below, but are not editable or named to edit once interview completes (attached)

2023-02-14T18:03:52.164Z CNTRLR « [Node 113] received information for parameter #1:
                                  parameter name:      LCD_invert
                                  value format:        UnsignedInteger
                                  value size:          1 bytes
                                  min value:           0
                                  max value:           1
                                  default value:       0
                                  is read-only:        false
                                  is advanced (UI):    false
                                  has bulk support:    true
                                  alters capabilities: false
[image]
[image]

AlCalzone, 2023-02-14:
@zwave-js-bot import config 1601 from zwa

AlCalzone, 2023-02-14:
Hm, automatically importing the config was not helpful. The IDs are a bit different than what we currently have for the Comet Z, but that can happen. We also don't have any config params defined at the moment, so they should be added too.

cadwizzard, 2023-02-14:
If it helps, this is a snippet from the interview log:

      "index": 0,
      "installerIcon": 4608,
      "userIcon": 4608,
      "status": 4,
      "ready": true,
      "isListening": false,
      "isRouting": true,
      "isSecure": false,
      "manufacturerId": 328,
      "productId": 3,
      "productType": 4,
      "firmwareVersion": "11.26.4",
      "zwavePlusVersion": 2

cadwizzard, 2023-03-14:
@AlCalzone Is there anything I can do to help get the config parameters added, so they can be edited in Home assistant?
I'm happy to help as I can't use these TRV's without the ability to alter some of the parameters (like invert display etc)

AlCalzone, 2023-03-14:
Yeah, see the comment above: https://github.com/zwave-js/node-zwave-js/issues/5464#issuecomment-1430241560

cadwizzard, 2023-03-14:
Thanks, I'll try to get what you need. 
I don't use JS UI, just the supported non UI zwave-js, so I dont believe I have access to parameter edit.

cadwizzard, 2023-03-14:
@kpine Thanks!
great workaround until the device gets proper support :)

bronger, 2023-08-02:
If I set manually in zwave-js-ui the Parameter 8 for the temperature offset, I get a “Parameter 8 set failed” in return. Is this because the device is not yet fully supported? I would have thought that setting parameters manually always works.

zwave-js-ui: 8.21.1.103c5be
zwave-js: 11.5.3

[image]

bronger, 2023-08-02:
Sorry, this was a mistake on my side. While the command still seems to be ignored, I cannot reproduce the error message any more. Probably I mixed up devices in the UI.

cadwizzard, 2023-09-17:
In case this info helps anyone who can make the config for this - I moved to JSUI as I couldn't get the parameters working otherwise .
[image]
`
2023-09-17T17:36:10.413Z - INTERVIEW STARTED
2023-09-17T17:36:10.472Z - INTERVIEW STAGE COMPLETED
Arg 0:
ProtocolInfo
2023-09-17T17:36:11.772Z - INTERVIEW STAGE COMPLETED
Arg 0:
NodeInfo
2023-09-17T17:36:11.827Z - VALUE ADDED
Arg 0:
└─commandClassName: Manufacturer Specific
└─commandClass: 114
└─property: manufacturerId
└─endpoint: 0
└─newValue: 328
└─propertyName: manufacturerId
2023-09-17T17:36:11.830Z - VALUE ADDED
Arg 0:
└─commandClassName: Manufacturer Specific
└─commandClass: 114
└─property: productType
└─endpoint: 0
└─newValue: 4
└─propertyName: productType
2023-09-17T17:36:11.833Z - VALUE ADDED
Arg 0:
└─commandClassName: Manufacturer Specific
└─commandClass: 114
└─property: productId
└─endpoint: 0
└─newValue: 3
└─propertyName: productId
2023-09-17T17:36:11.955Z - VALUE ADDED
Arg 0:
└─commandClassName: Version
└─commandClass: 134
└─property: libraryType
└─endpoint: 0
└─newValue: 3
└─propertyName: libraryType
2023-09-17T17:36:11.959Z - VALUE ADDED
Arg 0:
└─commandClassName: Version
└─commandClass: 134
└─property: protocolVersion
└─endpoint: 0
└─newValue: 7.15
└─propertyName: protocolVersion
2023-09-17T17:36:11.962Z - VALUE ADDED
Arg 0:
└─commandClassName: Version
└─commandClass: 134
└─property: firmwareVersions
└─endpoint: 0
└─newValue: 11.26, 1.4
└─propertyName: firmwareVersions
2023-09-17T17:36:11.965Z - VALUE ADDED
Arg 0:
└─commandClassName: Version
└─commandClass: 134
└─property: hardwareVersion
└─endpoint: 0
└─newValue: 1
└─propertyName: hardwareVersion
2023-09-17T17:36:13.044Z - VALUE ADDED
Arg 0:
└─commandClassName: Version
└─commandClass: 134
[... 231 lines cut ...]
└─newValue: 0
└─prevValue: 0
└─propertyName: alarmLevel
2023-09-17T17:36:17.658Z - VALUE ADDED
Arg 0:
└─commandClassName: Notification
└─commandClass: 113
└─property: System
└─propertyKey: Hardware status
└─endpoint: 0
└─newValue: 3
└─propertyName: System
└─propertyKeyName: Hardware status
2023-09-17T17:36:17.660Z - INTERVIEW STAGE COMPLETED
Arg 0:
CommandClasses
2023-09-17T17:36:17.663Z - INTERVIEW STAGE COMPLETED
Arg 0:
OverwriteConfig
2023-09-17T17:36:17.665Z - INTERVIEW STAGE COMPLETED
Arg 0:
Complete
2023-09-17T17:36:17.738Z - READY
2023-09-17T17:36:17.758Z - INTERVIEW COMPLETED
`

AlCalzone, 2023-09-21:
> (for some reason 'add code' isnt formatting the below like it should)

It's a bit counter-intuitive. First paste code, then select it, then press "format as code".

RickieCZ, 2023-09-26:
Product desc : [https://manual.zwave.eu/backend/make.php?lang=en&sku=EURECOMET&cert=]

cadwizzard, 2023-10-11:
Thanks to everyone that contributed to adding this ;)

Mart124, 2024-01-04:
@cadwizzard did you manage to tune parameters of your Comet devices ?
We fail [here](https://github.com/home-assistant/core/issues/103254), so would be pleased to have your report.
Thank you 👍

cadwizzard, 2024-01-04:
Hi.
I gave up I'm afraid, and moved to JSUI.
But I'm really unhappy with the CometZ.

Max reporting detail is 0.5c, unlike the Spirit (I have set to 0.3 normally)

The freeze setpoint is higher than the spirit

Also the spirits actually report detail to 0.01 degree, unlike the comet which is 0.5

Worst of all though is that they seemingly don't report temperature change sometimes reliably. I've had to request a multilevel sensor update to get the right info. This has left my boiler running for hours because the room temperature isn't reported.
I found a re-interview seems to fix that for a while and then they report every 0.5 degree again.

I had 20+ spirits and never had these issues. Upgraded 6 to CometZ. Had the problems on every one. Now can't trust them. Have returned 2 and now am trialling the ZigBee Bosch radiator thermostat 2 in z2mqqt. So far they have been brilliant. I didn't want to move away from z wave, but wanted to modernize the radiators.

I feel like Eurotronic have cheaped out with the hardware on the CometZ compared to the spirits.

Hope you have better luck.

Mart124, 2024-01-04:
Thank you for both your reports.
I plan to use them in Manufacturer specific mode (hence #6583), combined with a room-temperature sensor, and a smart Home Assistant thermostat (for ex. https://github.com/KartoffelToby/better_thermostat). So fully getting rid of the builtin temperature sensor and logic. Should give better results.

But for now stuck with parameters' configuration :|

cadwizzard, 2024-01-04:
> But for now stuck with parameters' configuration :|

.... So was I. The only reason I moved to JSUI was to accommodate these devices and the reporting resolution and mirror parameters.

The other 50+ devices were well supported in the native z wave js implementation

But I've found the network map and device management super useful. Especially because of the weirdness I was trying to figure out with the CometZ.

It's such a shame they don't work right.
Also if you change the display to mirrored, the up is setpoint down and down is up as you look at the TRV and it doesn't switch.
The spirits don't have that issue

Mart124, 2024-01-05:
OK, parameters issues fixed in https://github.com/zwave-js/node-zwave-js/pull/6583 👍 
May then be usable now 😉

PR #6336 (2023-10-11): feat(config): add Eurotronic Comet Z (700 series)
This is intended to close #5464

PR description here
https://[manual.zwave.eu/backend/make.php?lang=en&sku=EURECOMET&cert=](https://manual.zwave.eu/backend/make.php?lang=en&sku=EURECOMET&cert=)