zwave-js/zwave-js #5477: Climate control doens't work as expected ( decimals? )

rubentest, 2023-02-17:
Device AZAI6WSPFU2 doens't work as expected, isnt able to send int numbers to the zwave device.

Seems with decimals it's working.

Config of device:
https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x044e/AZAI6WSPFU2.json

Attached the log with the re-interwiew and testing with some params:

1rst attempt:  22ºC -> seems to send 22 and doens't work
2nd attempt: 22.0ºC -> seems to send 22  and doesn't work
3er attempt: 22,1ºC -> work but the unit returns 22

[zwavejs_current.log](https://github.com/zwave-js/node-zwave-js/files/10768124/zwavejs_current.log)

Many thanks!

AlCalzone, 2023-02-20:
@zwave-js-bot add compat flag to `0x044e:0x0004:0x0001`
```jsonc
		// This device only accepts Thermostat Setpoint Set commands with a precision of 1
		"overrideFloatEncoding": {
			"precision": 1
		}
```

AlCalzone, 2023-02-20:
@rubentest Can you please test [this file](https://raw.githubusercontent.com/zwave-js/node-zwave-js/add-compat-flag-1676904296/packages/config/config/devices/0x044e/AZAI6WSPFU2.json) as described [here](https://zwave-js.github.io/zwave-js-ui/#/device-support/custom-device-files)?

rubentest, 2023-02-21:
@AlCalzone worked perfectly without need to re-interwiew.
Many thanks

closed 2023-02-21 by AlCalzone

janneman001, 2023-11-13:
Hi,

I have exactly the opposite. I have 2 types of TRV 

Devolo (Danfoss) and Eurotronics Spirit. Both capable of 1 decimal in the setpoint. 

However both get the integer value. From the zwavejsui vue ui as well as from the websocket setValue. The vue ui represents the value correctly but the devices receive the integer value.

i.e. I send 14.1 via node => websocket, vue shows 14,1 which is correct by my language settings in the browser however the trv receives 14.

Or is the setting:

 "overrideFloatEncoding": {
			"precision": 1
		}

set for every trv now? 

If needed i can open a new issue for this. 
Regards Jan

PR #5483 (2023-02-21): fix(config): override setpoint precision for Airzone Aidoo Control HVAC unit
fixes: #5477

**TODO:**
- [x]	Change PR title to be more specific