zwave-js/zwave-js #5479: Go-Control (Linear) NGD00Z-4 Missing attributes

roblandry, 2023-02-19:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

Seems to be missing battery as reported in #4229.

I have reinterviewed the node multiple times.

Looking back at OG zwave configuration files, i see attributes that are not populated in zwave-js. The specific value for the battery being commandClass 128.
```xml
<CommandClass id="128" name="COMMAND_CLASS_BATTERY" version="1" request_flags="5" after_mark="true" issecured="true">
	<Instance index="1" />
	<Value type="byte" genre="user" instance="1" index="0" label="Battery Level" units="%" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="100" />
</CommandClass>
```

<details>
  <summary>Image from zwave-js</summary>

[image]

</details>

<details>
  <summary>Image from home assistant</summary>

[image]

</details>

<details>
  <summary>Config from zwave-js</summary>

  ```json
{
  "id": 5,
  "name": "Garage Door Opener",
  "loc": "",
  "values": [
    {
      "id": "5-102-0-signalingState-1",
      "nodeId": 5,
      "toUpdate": false,
      "commandClass": 102,
      "commandClassName": "Barrier Operator",
      "endpoint": 0,
      "property": "signalingState",
      "propertyName": "signalingState",
      "propertyKey": 1,
      "propertyKeyName": "1",
      "type": "number",
      "readable": true,
      "writeable": true,
      "label": "Signaling State (Audible)",
      "stateless": false,
      "commandClassVersion": 0,
      "min": 0,
      "max": 255,
[... 2022 lines cut ...]
`node-zwave-js` branch: 10.10.0
`zwave-js-ui` branch: 8.8.5

Using manually built version to update to [latest](https://github.com/hassio-addons/addon-zwave-js-ui/pull/463#issuecomment-1435756952).

### Did you change anything?

yes (please describe)

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Open zwave i believe, its been a while.

There are many comments about it not working in a long time.
As example:
https://community.home-assistant.io/t/issues-with-ngd00z-4-garage-door-opener/283517/23

### Attach Driver Logfile

S0_LEGACY Key emailed.
[zwavejs_2023-02-19.log](https://github.com/zwave-js/node-zwave-js/files/10776998/zwavejs_2023-02-19.log)

AlCalzone, 2023-02-20:
Unfortunately, I don't see a device interview in the log you posted. Please re-interview the device and post a log of that.

roblandry, 2023-02-21:
Updated OP and added todays device interview. In fact there are at least 2 interviews as it seemed sluggish so i pinged it during. So after complete i reinterviewed to ensure it was clean.

AlCalzone, 2023-02-22:
@roblandry you may have to try that again. I'm not sure if it is caused by node 5, but all of the interview attempts are during a phase where the stick refuses to send (probably due to excessive noise).

```
2023-02-21T22:22:46.952Z DRIVER « [REQ] [SendDataBridge]
                                    callback id:                           79
                                    transmit status:                       Fail, took 0 ms
...
```
appears in the log when that happens.

Might also be the 7.19 firmware. During one of the interview attempts, I see this flood of incoming reports from the node (which can cause the stick to refuse sending for a while):
```
2023-02-21T22:17:17.909Z SERIAL « 0x010c004984050604400772985ace                                      (14 bytes)
2023-02-21T22:17:17.911Z SERIAL » [ACK]                                                                   (0x06)
2023-02-21T22:17:17.912Z DRIVER « [Node 005] [REQ] [ApplicationUpdateRequest]
                                    payload: 0x050604400772985a
2023-02-21T22:17:17.972Z SERIAL « 0x010c004984050604400772985ace                                      (14 bytes)
2023-02-21T22:17:17.972Z SERIAL » [ACK]                                                                   (0x06)
2023-02-21T22:17:17.973Z DRIVER « [Node 005] [REQ] [ApplicationUpdateRequest]
                                    payload: 0x050604400772985a
2023-02-21T22:17:18.036Z SERIAL « 0x010c004984050604400772985ace                                      (14 bytes)
2023-02-21T22:17:18.037Z SERIAL » [ACK]                                                                   (0x06)
2023-02-21T22:17:18.038Z DRIVER « [Node 005] [REQ] [ApplicationUpdateRequest]
                                    payload: 0x050604400772985a
2023-02-21T22:17:18.169Z SERIAL « 0x010c004984050604400772985ace                                      (14 bytes)
2023-02-21T22:17:18.170Z SERIAL » [ACK]                                                                   (0x06)
2023-02-21T22:17:18.171Z DRIVER « [Node 005] [REQ] [ApplicationUpdateRequest]
                                    payload: 0x050604400772985a
2023-02-21T22:17:18.218Z SERIAL « 0x010c004984050604400772985ace                                      (14 bytes)
2023-02-21T22:17:18.219Z SERIAL » [ACK]                                                                   (0x06)
2023-02-21T22:17:18.220Z DRIVER « [Node 005] [REQ] [ApplicationUpdateRequest]
                                    payload: 0x050604400772985a
2023-02-21T22:17:18.306Z SERIAL « 0x010c004984050604400772985ace                                      (14 bytes)
2023-02-21T22:17:18.307Z SERIAL » [ACK]                                                                   (0x06)
2023-02-21T22:17:18.308Z DRIVER « [Node 005] [REQ] [ApplicationUpdateRequest]
                                    payload: 0x050604400772985a
2023-02-21T22:17:18.487Z SERIAL « 0x010c004984050604400772985ace                                      (14 bytes)
2023-02-21T22:17:18.488Z SERIAL » [ACK]                                                                   (0x06)
2023-02-21T22:17:18.489Z DRIVER « [Node 005] [REQ] [ApplicationUpdateRequest]
                                    payload: 0x050604400772985a
2023-02-21T22:17:18.586Z SERIAL « 0x010c004984050604400772985ace                                      (14 bytes)
2023-02-21T22:17:18.587Z SERIAL » [ACK]                                                                   (0x06)
2023-02-21T22:17:18.588Z DRIVER « [Node 005] [REQ] [ApplicationUpdateRequest]
                                    payload: 0x050604400772985a
2023-02-21T22:17:19.489Z SERIAL « 0x010c004984050604400772985ace                                      (14 bytes)
2023-02-21T22:17:19.491Z SERIAL » [ACK]                                                                   (0x06)
2023-02-21T22:17:19.492Z DRIVER « [Node 005] [REQ] [ApplicationUpdateRequest]
                                    payload: 0x050604400772985a
2023-02-21T22:17:19.518Z SERIAL « 0x010c004984050604400772985ace                                      (14 bytes)
2023-02-21T22:17:19.519Z SERIAL » [ACK]                                                                   (0x06)
2023-02-21T22:17:19.520Z DRIVER « [Node 005] [REQ] [ApplicationUpdateRequest]
                                    payload: 0x050604400772985a
2023-02-21T22:17:19.546Z SERIAL « 0x010c004984050604400772985ace                                      (14 bytes)
2023-02-21T22:17:19.547Z SERIAL » [ACK]                                                                   (0x06)
2023-02-21T22:17:19.548Z DRIVER « [Node 005] [REQ] [ApplicationUpdateRequest]
                                    payload: 0x050604400772985a
2023-02-21T22:17:19.679Z SERIAL « 0x010c004984050604400772985ace                                      (14 bytes)
2023-02-21T22:17:19.679Z SERIAL » [ACK]                                                                   (0x06)
2023-02-21T22:17:19.680Z DRIVER « [Node 005] [REQ] [ApplicationUpdateRequest]
                                    payload: 0x050604400772985a
2023-02-21T22:17:19.728Z SERIAL « 0x010c004984050604400772985ace                                      (14 bytes)
2023-02-21T22:17:19.729Z SERIAL » [ACK]                                                                   (0x06)
2023-02-21T22:17:19.730Z DRIVER « [Node 005] [REQ] [ApplicationUpdateRequest]
                                    payload: 0x050604400772985a
[... cut ...]
2023-02-21T22:20:35.062Z SERIAL » [ACK]                                                                   (0x06)
2023-02-21T22:20:35.066Z DRIVER « [REQ] [SerialAPIStarted]
                                    wake up reason:        SoftwareReset
                                    watchdog enabled:      false
                                    generic device class:  0x02
                                    specific device class: 0x01
                                    always listening:      false
                                    supports Long Range:   false
```

roblandry, 2023-02-23:
Ok, new logfile. I now see what you mean with all the resets. I moved the dongle to a usb 2.0 hub on a long extension and seems to be much better (less noise), yet still missing the attributes.

AlCalzone, 2023-02-24:
Ok the issue is that the device doesn't report support for the Battery CC. Neither insecurely:
```
2023-02-23T23:02:43.572Z CNTRLR « [Node 005] node info received
                                  supported CCs:
                                  · Manufacturer Specific
                                  · Security
                                  · Device Reset Locally
```
nor securely:
```
2023-02-23T23:02:43.802Z CNTRLR « [Node 005] received secure commands (S0)
                                  supported CCs:
                                  · Version
                                  · Barrier Operator
                                  · Application Status
                                  · Notification
                                  · Association
```
Looking at the OZW file there are probably a few more CCs missing, but I don't trust that file (why does it contain thermostat related CCs?).
I'll post a PR with an update config to force-add the Battery CC and have you test it.

AlCalzone, 2023-02-24:
Ok @roblandry please test [this file](https://raw.githubusercontent.com/zwave-js/node-zwave-js/NGD00Z-4-battery/packages/config/config/devices/0x014f/ngd00z-4.json) as described [here](https://zwave-js.github.io/zwave-js-ui/#/device-support/custom-device-files).
You will have to re-interview again for this to get picked up.

roblandry, 2023-02-24:
Ok, the new info is populated but providing unknown values. In further reading today, i am not sure it has ever actually provided an actual correct value. I only recall actually seeing 100%, despite the battery being in for years. I think it may be safe to close this issue and remove the PR.

closed 2023-03-06 by AlCalzone