zwave-js/zwave-js #5485: Removing failed nodes does not always remove the node from the UI

ColColonCleaner, 2023-02-21:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

8.8.3

### ZwaveJS version

10.5.5

### Describe the bug

Removing nodes doesn't always remove the node from the UI. Requires a VM bounce to remove it. Subsequent removals of the node fail saying the node doesn't exist but the row remains displayed.

### To Reproduce

Remove more than one node from a network. Screenshot attached.

### Expected behavior

The node row is removed when the node is removed from the network.

AlCalzone, 2023-02-23:
Can reproduce. Node 13 was removed immediately, Node 14 did not get removed from the UI until reloading the page:

```
2023-02-23 13:08:39.416 DEBUG SOCKET: Event ZWAVE_API emitted to ic4HlA8pRHqTKdOlAAAH
2023-02-23 13:08:39.422 INFO Z-WAVE: Calling api removeFailedNode with args: [ 13, [length]: 1 ]
13:08:39.422 CNTRLR » [Node 013] pinging the node...
13:08:39.430 SERIAL » 0x010d00a9010d010025000000000477                                    (15 bytes)
13:08:39.430 DRIVER » [Node 013] [REQ] [SendDataBridge]
                      │ source node id:   1
                      │ transmit options: 0x25
                      │ callback id:      4
                      └─[NoOperationCC]
13:08:39.437 SERIAL « [ACK]                                                                   (0x06)
13:08:39.438 SERIAL « 0x010401a90152                                                       (6 bytes)
13:08:39.439 SERIAL » [ACK]                                                                   (0x06)
13:08:39.440 DRIVER « [RES] [SendDataBridge]
                        was sent: true
13:08:42.849 SERIAL « 0x011d00a90401014f007f7f7f7f7f00010600000000420300007f7f7f7f7f46    (31 bytes)
13:08:42.850 SERIAL » [ACK]                                                                   (0x06)
13:08:42.851 DRIVER « [REQ] [SendDataBridge]
                        callback id:            4
                        transmit status:        NoAck, took 3350 ms
                        routing attempts:       3
                        protocol & route speed: Z-Wave, 40 kbit/s
                        TX channel no.:         1
                        beam:                   1000 ms
13:08:42.854 CNTRLR   [Node 013] The node did not respond after 1 attempts, it is presumed dead
13:08:42.856 CNTRLR   [Node 013] ping failed: Failed to send the command after 1 attempts (Status No
                      Ack) (ZW0204)
13:08:42.858 SERIAL » 0x010500610d0593                                                     (7 bytes)
13:08:42.858 DRIVER » [REQ] [RemoveFailedNode]
                        payload: 0x0d05
13:08:42.866 SERIAL « [ACK]                                                                   (0x06)
13:08:42.867 SERIAL « 0x01040161009b                                                       (6 bytes)
13:08:42.867 SERIAL » [ACK]                                                                   (0x06)
13:08:42.868 DRIVER « [RES] [RemoveFailedNode]
                        payload: 0x00
13:08:46.311 SERIAL « 0x0105006105019f                                                     (7 bytes)
13:08:46.312 SERIAL » [ACK]                                                                   (0x06)
13:08:46.313 DRIVER « [REQ] [RemoveFailedNode]
                        payload: 0x0501
13:08:46.318 CNTRLR   [Node 013] [-] [Security 2] interviewComplete (was tru [Endpoint 0] [internal]
                      e)
13:08:46.319 CNTRLR   [Node 013] [-] [Manufacturer Specific] interviewComple [Endpoint 0] [internal]
                      te (was true)
13:08:46.320 CNTRLR   [Node 013] [-] [Version] interviewComplete (was true)  [Endpoint 0] [internal]
13:08:46.320 CNTRLR   [Node 013] [-] [Z-Wave Plus Info] zwavePlusVersion (wa [Endpoint 0] [internal]
                      s 1)
13:08:46.321 CNTRLR   [Node 013] [-] [Z-Wave Plus Info] nodeType (was 0)     [Endpoint 0] [internal]
13:08:46.321 CNTRLR   [Node 013] [-] [Z-Wave Plus Info] roleType (was 7)     [Endpoint 0] [internal]
13:08:46.321 CNTRLR   [Node 013] [-] [Z-Wave Plus Info] installerIcon (was 7 [Endpoint 0] [internal]
                      68)
13:08:46.322 CNTRLR   [Node 013] [-] [Z-Wave Plus Info] userIcon (was 768)   [Endpoint 0] [internal]
13:08:46.322 CNTRLR   [Node 013] [-] [Z-Wave Plus Info] interviewComplete (w [Endpoint 0] [internal]
                      as true)
2023-02-23 13:08:46.325 INFO Z-WAVE: [Node 013] Removed
2023-02-23 13:08:46.325 INFO Z-WAVE: [Node 013] Removed
2023-02-23 13:08:46.325 DEBUG Z-WAVE: Updating store nodes.json
2023-02-23 13:08:46.326 INFO Z-WAVE: Success zwave api call removeFailedNode undefined
2023-02-23 13:08:58.095 DEBUG SOCKET: Event ZWAVE_API emitted to ic4HlA8pRHqTKdOlAAAH
2023-02-23 13:08:58.095 INFO Z-WAVE: Calling api removeFailedNode with args: [ 14, [length]: 1 ]
13:08:58.096 CNTRLR » [Node 014] pinging the node...
13:08:58.102 SERIAL » 0x010d00a9010e010025000000000676                                    (15 bytes)
13:08:58.103 DRIVER » [Node 014] [REQ] [SendDataBridge]
                      │ source node id:   1
                      │ transmit options: 0x25
                      │ callback id:      6
                      └─[NoOperationCC]
13:08:58.108 SERIAL « [ACK]                                                                   (0x06)
13:08:58.109 SERIAL « 0x010401a90152                                                       (6 bytes)
[... 385 lines cut ...]
2023-02-23 13:09:02.531 INFO Z-WAVE: [Node 014] ValueId 135-0-80-5 removed
2023-02-23 13:09:02.531 INFO Z-WAVE: [Node 014] Value removed: 135-0-80-5
13:09:02.531 CNTRLR   [Node 014] [Indicator] 80[5]: metadata updated                    [Endpoint 0]
2023-02-23 13:09:02.532 INFO Z-WAVE: [Node 014] Metadata updated: 135-0-80-5
13:09:02.532 CNTRLR   [Node 014] [-] [Basic] currentValue (was 0)                       [Endpoint 0]
2023-02-23 13:09:02.532 WARN Z-WAVE: [Node 014] ValueId 32-0-currentValue removed: no such node
2023-02-23 13:09:02.532 INFO Z-WAVE: [Node 014] Value removed: 32-0-currentValue
[... cut ...]
13:09:02.534 CNTRLR   [Node 014] [-] [Notification] Weather Alarm[unknown] (was 6)      [Endpoint 0]
2023-02-23 13:09:02.534 INFO Z-WAVE: [Node 014] ValueId 113-0-Weather Alarm-unknown removed
2023-02-23 13:09:02.534 INFO Z-WAVE: [Node 014] Value removed: 113-0-Weather Alarm-unknown
2023-02-23 13:09:02.535 INFO Z-WAVE: [Node 014] Removed
2023-02-23 13:09:02.535 INFO Z-WAVE: [Node 014] Removed
2023-02-23 13:09:02.535 DEBUG Z-WAVE: Updating store nodes.json
2023-02-23 13:09:02.536 INFO Z-WAVE: Success zwave api call removeFailedNode undefined
```

AlCalzone, 2023-02-23:
Interestingly, while Node 14 did not get removed, Node 15 was missing from the nodes table, although I did not remove it. After reloading the page, Node 14 vanished and 15 came back.