zwave-js/zwave-js #5490: Multicast not working with Qubino Mini Dimmer

nmeachen, 2023-02-23:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

I am trying to use the zwavejs multicast service call to turn on two lights simultaneously, Homeassistant (2023.2.5) shows them on, the ZwaveJS2MQTT (8.8.6) control panel shows them as on, but the lights do not turn on. The lights are Qubino Mini Dimmers (v21.6). Is multicast part of the Zwave specification or something ZwaveJS (10.10.0) implements? Does it require certain functionality in the device?

The only suspicious thing I see in the log is
   2023-02-23T03:00:15.840Z DRIVER   dropping CC with invalid values (Reason: Unsupported rate type Unspecified or corrupted data)

Unfortunately, repeating the tests does not produce this error, so that message is probably just due to something else on the ZWave network.

service: zwave_js.multicast_set_value
data:
  command_class: '37'
  endpoint: 0
  property: targetValue
  value: true
target:
  entity_id:
    - light.living_room_dimmer_left
    - light.living_room_dimmer_right

### Device information

Manufacturer: Qubino
Model name: Mini Dimmer ZMNHHD
Node ID in your network: 7 & 10

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:
`zwave-js-ui` branch:

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

The relevant activity in the log starts at 2023-02-23T16:00:23.597Z, a multicast turn on for nodes 10 & 7 followed by the individual turn offs through the HA UI. Then an individual turn on for node 7 through the HA UI.

The multicast does not seem to result in any communication with the nodes.

[zwavejs_current.log.gz](https://github.com/zwave-js/node-zwave-js/files/10815772/zwavejs_current.log.gz)

AlCalzone, 2023-02-23:
The nodes are using Security S2, for which multicast is not implemented in Z-Wave JS **yet**.
Support is coming very soon though: https://github.com/zwave-js/node-zwave-js/pull/5475

nmeachen, 2023-02-23:
Is Multicast part of the Z-Wave specification or something Z-Wave JS facilitates?

AlCalzone, 2023-02-24:
It is part of the specification, and relatively complicated for S2.

closed 2023-03-07 by AlCalzone

nmeachen, 2023-03-09:
Sadly the S2 changes for multicast have not got the Qubino dimmers working for me.

```
2023-03-09T01:28:16.912Z SERIAL » 0x010c0014020a07032501ff251a0f                                      (14 bytes)
2023-03-09T01:28:16.915Z DRIVER » [REQ] [SendDataMulticast]
                                  │ target nodes:     10, 7
                                  │ transmit options: 0x25
                                  │ callback id:      26
                                  └─[BinarySwitchCCSet]
                                      target value: true
2023-03-09T01:28:16.925Z SERIAL « [ACK]                                                                   (0x06)
2023-03-09T01:28:16.928Z SERIAL « 0x0104011401ef                                                       (6 bytes)
2023-03-09T01:28:16.930Z SERIAL » [ACK]                                                                   (0x06)
2023-03-09T01:28:16.933Z DRIVER « [RES] [SendDataMulticast]
                                    was sent: true
2023-03-09T01:28:17.020Z SERIAL « 0x010500141a00f4                                                     (7 bytes)
2023-03-09T01:28:17.021Z SERIAL » [ACK]                                                                   (0x06)
2023-03-09T01:28:17.022Z DRIVER « [REQ] [SendDataMulticast]
                                    callback id:     26
                                    transmit status: OK
2023-03-09T01:28:17.027Z CNTRLR   [Node 010] [~] [Binary Switch] currentValue: false => true        [Endpoint 0]
2023-03-09T01:28:17.030Z CNTRLR   [Node 007] [~] [Binary Switch] currentValue: false => true        [Endpoint 0]
2023-03-09T01:28:17.034Z CNTRLR   [Node 010] [~] [Binary Switch] targetValue: false => true         [Endpoint 0]
2023-03-09T01:28:17.037Z CNTRLR   [Node 007] [~] [Binary Switch] targetValue: false => true         [Endpoint 0]
```

The switch entities show as ON in ZwaveJS and HomeAssistant, but they do not turn on.

AlCalzone, 2023-03-09:
How did you trigger the multicast? I think the applications aren't using the new methods yet.

nmeachen, 2023-03-10:
I used the zwave_js multicast service in HA 2023.3.2.

```
service: zwave_js.multicast_set_value
data:
  command_class: 37
  endpoint: 0
  property: targetValue
  value: true
target:
  entity_id:
    - switch.living_room_dimmer_left
    - switch.living_room_dimmer_right
```