zwave-js/zwave-js #5496: DRIVER dropping CC with invalid values

ridderr, 2023-02-25:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

Many log entries as shown below. Unfortunately, I'm not able to increase loglevel because it only seems to accept Info.
The Z-wave nodes seems to operate without error but these logentries are coming in constantly.

2023-02-25T09:50:01.652Z DRIVER dropping CC with invalid values (Reason: Unexpected meter type or corrupted data)
2023-02-25T09:50:04.649Z DRIVER dropping CC with invalid values (Reason: Unexpected meter type or corrupted data)
2023-02-25T09:50:07.649Z DRIVER dropping CC with invalid values (Reason: Unexpected meter type or corrupted data)
2023-02-25T09:50:10.658Z DRIVER dropping CC with invalid values (Reason: Unexpected meter type or corrupted data)
2023-02-25T09:50:13.650Z DRIVER dropping CC with invalid values (Reason: Unexpected meter type or corrupted data)
2023-02-25T09:50:16.647Z DRIVER dropping CC with invalid values (Reason: Unexpected meter type or corrupted data)
2023-02-25T09:50:19.649Z DRIVER dropping CC with invalid values (Reason: Unexpected meter type or corrupted data)
2023-02-25T09:50:22.650Z DRIVER dropping CC with invalid values (Reason: Unexpected meter type or corrupted data)
2023-02-25T09:50:25.649Z DRIVER dropping CC with invalid values (Reason: Unexpected meter type or corrupted data)
2023-02-25T09:50:28.655Z DRIVER dropping CC with invalid values (Reason: Unexpected meter type or corrupted data)
2023-02-25T09:50:31.649Z DRIVER dropping CC with invalid values (Reason: Unexpected meter type or corrupted data)

### Device information

No sure which device is causing this.

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

Z-Wave JS Current version: 0.1.76
Driver Version: 10.10.0
Server Version: 1.26.0
Home ID: 3610736349

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### If yes, where did it work?

Is not related to a new device. I could be due to a new version of HA.

### Attach Driver Logfile

2023-02-25T09:50:01.652Z DRIVER dropping CC with invalid values (Reason: Unexpected meter type or corrupted data)
2023-02-25T09:50:04.649Z DRIVER dropping CC with invalid values (Reason: Unexpected meter type or corrupted data)
2023-02-25T09:50:07.649Z DRIVER dropping CC with invalid values (Reason: Unexpected meter type or corrupted data)
2023-02-25T09:50:10.658Z DRIVER dropping CC with invalid values (Reason: Unexpected meter type or corrupted data)
2023-02-25T09:50:13.650Z DRIVER dropping CC with invalid values (Reason: Unexpected meter type or corrupted data)
2023-02-25T09:50:16.647Z DRIVER dropping CC with invalid values (Reason: Unexpected meter type or corrupted data)
2023-02-25T09:50:19.649Z DRIVER dropping CC with invalid values (Reason: Unexpected meter type or corrupted data)
2023-02-25T09:50:22.650Z DRIVER dropping CC with invalid values (Reason: Unexpected meter type or corrupted data)
2023-02-25T09:50:25.649Z DRIVER dropping CC with invalid values (Reason: Unexpected meter type or corrupted data)
2023-02-25T09:50:28.655Z DRIVER dropping CC with invalid values (Reason: Unexpected meter type or corrupted data)
2023-02-25T09:50:31.649Z DRIVER dropping CC with invalid values (Reason: Unexpected meter type or corrupted data)

AlCalzone, 2023-02-25:
Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` and attach it here as a file (drag & drop into the text field).

While the log is running, re-interview the device. If that doesn't fix the issue, post the interview log and one where I can see the dropped commands.

ridderr, 2023-02-26:
Thanks for you feedback. I had to move over to Z-Wave JS UI (was running Z-Wave JS).
Problem has been solved and what an improvement with this Add-on. Great job.

closed 2023-02-26 by ridderr

AlCalzone, 2023-02-26:
It's the same driver underneath, but the don't share a cache, so Z-Wave UI probably interviewed the device again, which solved the problem.