zwave-js/zwave-js #5497: TZ69E not working correctly

glendower, 2023-02-25:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug? The TZ69E is included but does not operate as expected. The energy values are not updated and the switch operation is not reflected on the device.

What do you observe?

What did you expect to happen? Energy values should be changing as power is consumed and the switch should work.

Steps to reproduce the behavior:
  1. Go to '...'
  2. Click on '...'
  3. Scroll down to '...'
  4. See error

### Device information

Manufacturer: TKBHome
Model name: TZ69E
Node ID in your network: 2

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 1.6.3
`zwave-js-ui` branch: 8.8.6

### Did you change anything?

yes (please describe)

### If yes, what did you change?

nothing

### Did this work before?

Don't know, this is a new device

### If yes, where did it work?

Worked perfectly on Vera

### Attach Driver Logfile

[zwavejs_current (1).log](https://github.com/zwave-js/node-zwave-js/files/10831984/zwavejs_current.1.log)

closed 2023-03-19 by zwave-js-assistant[bot]

mullermn, 2023-10-24:
Thanks for taking a look at this.
[zwavejs_2023-10-24.log](https://github.com/zwave-js/node-zwave-js/files/13112414/zwavejs_2023-10-24.log)

I changed the reporting period for Watts down to 3 (15 seconds) at 2023-10-24T08:26:45.143Z, so the reports start there.

I am open to the idea that this device has malfunctioned, but it's not my first instinct. It spent the entire time since it was working plugged in to the same indoor socket, safe from any damage, and all its other functions seem to work fine - seems a bit odd for this specific function to have stopped working on a hardware level.

AlCalzone, 2023-10-24:
Seems to be working fine now, right? I'd lower the reporting period though, too much energy reporting can easily bring a Z-Wave network to its knees.

> I am open to the idea that this device has malfunctioned, but it's not my first instinct. It spent the entire time since it was working plugged in to the same indoor socket, safe from any damage, [...]

These reports are unsolicited reports, initiated by the device. The controller / Z-Wave JS simply receives them. So it's either something with the device, or the connectivity is bad enough for reports to get lost. This doesn't look too healthy:
```
transmit status:        OK, took 220 ms
repeater node IDs:      15
routing attempts:       4
protocol & route speed: Z-Wave, 9.6 kbit/s
routing scheme:         NLWR
ACK RSSI:               -92 dBm
```
...considering that it started out with 100 kbps and a direct route:
```
transmit status:        OK, took 10 ms
routing attempts:       1
protocol & route speed: Z-Wave, 100 kbit/s
routing scheme:         LWR
ACK RSSI:               -95 dBm
```
... and the RSSI values are only 10 dBm above background noise levels.

You may want to take a look at these links to try and improve this:
https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting
and
https://zwave-js.github.io/node-zwave-js/#/troubleshooting/network-health?id=testing-the-connection-strength

mullermn, 2023-10-24:
`Seems to be working okay now, right? I'd lower the reporting period though, too much energy reporting can easily bring a Z-Wave network to its knees.`
Yeah, I only put it up that high for testing. It's not working though - the reports come through, but ever single one has a zero value (this is with a device plugged in to the switch and turned on - a lamp with an incandescent bulb, so it should be drawing enough power to register).

The weird route could be my fault - I've moved the device to my desk while I'm working on it and that happens to be next to the zwave radio. Its normal location is on the other side of the house, so if it's trying to relay via its normal route then the signal would probably be pretty weak. I'll do a heal on it and see if that helps.

If a weak signal was the cause then surely the reports wouldn't get through at all rather than get through with the wrong value?

AlCalzone, 2023-10-24:
> If a weak signal was the cause then surely the reports wouldn't get through at all rather than get through with the wrong value?

I agree. You could try excluding, factory resetting and including the device again. Maybe that convinces it to send the correct measurements again. Other than that I have no idea what else you could do besides replacing it.

mullermn, 2023-10-24:
OK. I've tried every permutation of resetting, excluding etc that I can think of, so I think I'm going to have to call the hardware dead. Thanks for looking in to it.