zwave-js/zwave-js #5511: Missing device configuration: iDevices In-Wall Smart Dimmer

spatil47, 2023-03-01:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

iDevices In-Wall Smart Dimmer

### Manufacturer ID

0x0458

### Product Type

0x4944

### Product ID

0x3236

### Firmware Version

5.55

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/4319

### Do you have a manual?

https://products.z-wavealliance.org/ProductManual/File?folder=&filename=product_documents/4319/61303-1%20EnFrSp%20QSG%20v1.pdf

spatil47, 2023-03-02:
The Z-Wave Alliance product information for configuration parameter 16, parameter value 1 is incorrect: parameter value 1 does _not_ enable exclusion by double-tap-on-double-tap-off, but enables dimming by press-and-hold instead of switching-only.

AlCalzone, 2023-03-06:
@zwave-js-bot import config 4319 from zwa

PR #5521 (2024-07-09): feat(config): add iDevices In-Wall Smart Dimmer
fixes: #5511

**TODO:**
- [x] Change PR title to be more specific
- [x] Consider https://github.com/zwave-js/node-zwave-js/issues/5511?notification_referrer_id=NT_kwDOAQ0vDbM1NzQxNTg5NDUzOjE3NjQxMjI5#issuecomment-1452409435
closed 2024-07-09 by AlCalzone