zwave-js/zwave-js #5517: Doorsensor ZWA011 working incorrectly since last update

Maes152, 2023-03-02:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

**What causes the bug?**
I am using..
addon-zwave-js-ui v1.6.3 for HA 
This is zwave-js-ui v8.8.6
This is node-zwave-js v10.10.0
This is zwave-js-server v1.26.0

The bug started appearing since updating to this version. 

**What do you observe?**
The Aeotec window/door sensor ZWA011 is suddenly working incorrectly. Upon opening/closing the door the sensor is no longer updating the same entities. I used to use "access_control_window_door_is_open". This is now no longer updated. Instead a "new" entity exists called "Door state". This gives the value "Window/door is closed" on closing but gives a number "5632" when opening. 

**What did you expect to happen?**
Either the old state to continue working as it always has or the new entity to display the value "Window/door is open" instead of a number 5632.

**Steps to reproduce the behavior:**
  1. Open/close your door ;)

### Device information

Manufacturer: Aeotec
Model name: ZWA011
Node ID in your network: 41
Firmware: 2.0.19

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: main
`zwave-js-ui` branch: main

### Did you change anything?

yes (please describe)

### If yes, what did you change?

Updated to latest version as described above

### Did this work before?

Yes (please describe)

### If yes, where did it work?

It worked fine before update

### Attach Driver Logfile

Closing: 
`2023-03-02T05:58:33.787Z - VALUE UPDATED 
Arg 0:
└─commandClassName: Notification
└─commandClass: 113
└─property: Access Control
└─propertyKey: Door state
└─endpoint: 0
└─newValue: 23
└─prevValue: 5632
└─propertyName: Access Control
└─propertyKeyName: Door state`

Opening:
`2023-03-02T05:57:35.852Z - VALUE UPDATED 
Arg 0:
└─commandClassName: Notification
└─commandClass: 113
└─property: Access Control
└─propertyKey: Door state
└─endpoint: 0
└─newValue: 5632
└─prevValue: 23
└─propertyName: Access Control
└─propertyKeyName: Door state`

Maes152, 2023-03-02:
Thank you for responding. I should've mentioned that I already re-interviewed the node. 
This is what gave me the new entity called "Door state" (param 113). 
Although I am puzzled why this param 113 is not documented in the DB config I linked to, that is not the source of the issue.

Thank you for providing the links. The issue that was logged at the HA repo #88060 specifically mentions that the zwave integration is supposed to provide the label metadata. And so I believe this might not be setup correctly currently, otherwise I would not get the label "5632". Do you agree?

Maes152, 2023-03-02:
Will do. Thank you for your help. I will let you know if I am able to resolve the issue.

stickerbob, 2023-03-04:
I am having the same issues with my ZWA011.  I have re-interviewed it 4 times and excluded and included it again.  Nothing has solved it for me.  After a re-interview, the sensor reports closed upon the first closure, but nothing after that.  It does report Door/Window Sensor 7 Window/door is open in regular position as on.

stickerbob, 2023-03-04:
Can you explain or point me to the link?  I am new to Hass and Z-wave JS, so I am still learning.  I am sure it's something simple.  I appreciate your help.

stickerbob, 2023-03-04:
Ah, my fault.  Thanks again!

Maes152, 2023-03-06:
After a proper interview, my issue was resolved. Thank you @kpine 
Apparently I had to press a special button on the ZWA011 to accept the interview process rather than just triggering the open/close. 

Just wondering for next time how can I stay properly informed of changes such as these as they are being pushed through? I didn't read anything about it in the changelog.

Maes152, 2023-03-06:
Fair enough. This PR mentions that the labels for "Notification CC event parameters" are now supported. 
But it doesn't mention anything about the old entities not working / being updated anymore (parameter 3 in my OP).
Again I'd like to direct you to the [config DB](https://devices.zwave-js.io/?jumpTo=0x0371:0x0002:0x000b:0.0) that to this day still describes the "old" situation and makes no mention of the new binary sensor parameter 113. 

I've been using this sensor for quite some time and I would definitely consider whatever happened to be a breaking change.

AlCalzone, 2023-03-06:
> But it doesn't mention anything about the old entities not working

It does though 
> The Door/window state of devices which do send an enum value for the open state is possibly influenced by this though:
> 
> Before:
> 
> 22: Window/door is open
> 23: Window/door is closed
> After:
> 
> 22: Window/door is open (possibly unused)
> 23: Window/door is closed
> 5632: Window/door is open in regular position
> 5633: Window/door is open in tilt position

Maes152, 2023-03-06:
I see where you're coming from but the actual sensors have changed. Someone as inexperienced as myself who reads this changelog isn't immediately triggered by it being a breaking change.
In the current solution the "old entity" is simply not being updated anymore.

I will henceforth read the changelog with greater care however. I want to thank you both for your contributions and time you've put in to this awesome integration.

closed 2023-03-06 by AlCalzone