zwave-js/zwave-js #5519: Aeotec - MultiSensor 6 (ZW100) incorrect reporting.

Tamas-Toth-ebola, 2023-03-03:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

YES, BUT a Z-Wave JS UI developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

**What causes the bug?**

The device basically works **but it seems with incorrect state reports.**

Just bought my _(n.)_ Z-Wave device some days ago, an Aeotec - MultiSensor 6 _(ZW100)_, which after its setup/initialization seems to work fine _(I can see the sensor data, also can configure, etc. without any problem)_, but!.

From the start of the device usage, I had a problem, for which I had some days to realize its 'real background'. As a battery-operated _(battery or USB powered)_ device, when its state switch to `asleep`, Z-Wave JS UI sometimes show me that the node is `dead`. At first days I thought that that is really `dead`, but interestingly after I make some movement in front of it, it automatically 'woke up' and sent the corresponding sensor data. After a few times, its state showed on the UI again as `dead`, which could easily be changed to `alive` again _(on Z-Wave JS UI device summarization row)_ with just a movement in front of the motion sensor. At this point it was just interesting, I could not say what was in the background of it, but I feared from 'faulty device' as I never met such a situation.

The second thing I realized is that Z-Wave JS UI reporting wrongly its 'powering state'.
[image]
While it was on USB, it had been reported as 'mains-powered'. It's OK. But after I changed it to 'battery-powered', the state report was still stayed the same: 'mains-powered'. At the first time I thought that it is correlated to the problematic `dead` state, I mean, the device works somewhat faulty... but after a lot of experiments, I realized that basically there is no problem with the device. It is never in a real `dead` state, and also works fine with batteries, in fact, one of its parameters shows correctly that it is 'battery-powered', but the main state report _(on Z-Wave JS UI device summarization row)_ is wrong.
[image]

As you can see from the screen, its 'Configuration v1' parameter-set shows that the device is `battery-powered`, and also in an `asleep` state, while its main data row shows that `mains-powered` and `alive`. Interesting.

Sadly, currently, I can not send you a screen of a `dead` state, but if you tell me how and which data need to send you to investigate the situation, I could be more then happy to help.

_(Of course at the very end in Home Assistant I see the same, but the problem in in its roots, in this case in Z-Wave JS, as the problem is available independently from Home Assistant)_

**What do you observe?**

Detailed in the description.

**What did you expect to happen?**

Detailed in the description.

**Steps to reproduce the behavior:**

Just check the node information and see its misleading data.

### Device information

Manufacturer: Aeotec
Model name: ZW100
Node ID in your network: 5

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
[image]

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwave-js-ui-store.zip](https://github.com/zwave-js/node-zwave-js/files/10881078/zwave-js-ui-store.zip)

AlCalzone, 2023-03-03:
If you plan to operate the device on battery power, it needs to be included into the network on battery power too. Otherwise it will be identified as a mains-powered device per the Z-Wave protocol and be treated as such.

Tamas-Toth-ebola, 2023-03-03:
@AlCalzone 

I have other battery-powered devices with properly working...

[image]
[image]

...where the difference is only their inclusion methods, and those are SmartStart/provisioning ready devices compared to the 'current one'.

I theoretically understand what you wrote and really do not want to spend your time, but could you please help me with how can I make such a network, as for now, I had never _(I had never had to)_ done anything else over a simple inclusion/exclusion? Just a link with content to drive me into the right direction would be more than enough, but I'm not really seeing it on https://zwave-js.github.io/node-zwave-js/#/ site yet.

Thanks a lot for it!

AlCalzone, 2023-03-03:
- Exclude the device 
- Keep it on battery power, not USB 
- Include it again

Tamas-Toth-ebola, 2023-03-03:
Thank you, it works and hopefully will solve my mentioned problems. Now Z-Wave JS US shows that the device is battery-powered, and reports `asleep` state default. Also after manual wake-up, the state changed also the proper one.

Thanks again, and sorry for the trivial question. Please close the issue.

closed 2023-03-03 by AlCalzone