zwave-js/zwave-js #5540: Device interview, missing Binary Switch.

sheggy012, 2023-03-08:
zwave-js-ui: 8.9.0
zwave-js: 10.10.0

Hi Everyone, I finally made the move in my homeassitant to migrate from legacy openzwave to zwavejs ui. So far the migration has worked great. However, I have a device that I just can't get to run. 
The device is a: "EBV HeatApp! Floor" 
Product Manufacturer: 0x017c
Product Identifiers: 0x0006:0x0003

This is actually a simple binary switch on which you can switch 8 relays. Since the migration only one relay can be switched, the other 7 are either not shown at all or sometimes only shown as readable "Binary Switch vundefined" (see picture).

Is there any way how I can get this device fixed? Maybe disabling the interview and adding the 7 missing relays? 

[image]
[node_26.zip](https://github.com/zwave-js/zwave-js-ui/files/10919112/node_26.zip)
[zwave-js-ui-logs.zip](https://github.com/zwave-js/zwave-js-ui/files/10919113/zwave-js-ui-logs.zip)

I would be very happy about any help. The device controls my heating, and its currently snowing :-)

AlCalzone, 2023-03-08:
I can certainly add a workaround for this in the driver code, but this seems like you need some more urgent help. This will require editing the cache files while the driver is not connected to a stick.
I'm not sure if this will work, but you could try the following:
1. Open Z-Wave JS UI, remember your home ID (somewhere in the top right corner)
2. Go to the store view (in Z-UI), download the 3 `.jsonl` files which have this home ID in the name
3. Send me the files (e-mail in my profile), I'll edit in the other endpoints
4. When you have the files back, pull the stick, replace the files' contents in the store
5. Put the stick back in.
6. Maybe: refresh the values for the Binary switches

sheggy012, 2023-03-09:
Thank you very much. Your modified jsonl files have provided urgent relief. It now works as desired. If you want me to test something out here, I'm always happy to do so.

AlCalzone, 2023-07-10:
@sheggy012 this should be fixed in the upcoming release (needs re-interview, which would override your manual edits). If you're adventurous, give it a test please.

closed 2023-07-10 by zwave-js-bot