zwave-js/zwave-js #5546: Logic Group Matrix ZRB5120 issue

andlo, 2023-03-09:


robertsLando, 2023-03-09:
Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` and attach it here as a file (drag & drop into the text field).

andlo, 2023-03-09:
Great. I made a logfile as described, and made a reinterviev for node 40. Then I turned on and og the to switches (in Homeassistant) pressed Ping and turned on and of again.
And then attatched the logfile here :) 
[zwavejs_2023-03-09.log](https://github.com/zwave-js/node-zwave-js/files/10932340/zwavejs_2023-03-09.log)

AlCalzone, 2023-03-09:
I'm not seeing anything related to turning on or off in the driver log. The rest looks okay. This might be one of the cases where we actually need logs from Z-Wave JS UI ("Application Logs").

Just capture the failing interaction in those.

andlo, 2023-03-09:
[z-ui_2023-03-09.log](https://github.com/zwave-js/node-zwave-js/files/10932719/z-ui_2023-03-09.log)
@AlCalzone This log ?

andlo, 2023-03-09:
I enabled the log to file and debug. Then did a turn on eand of the to entities in Homeassistant, pressed Ping and turn on and off again on the to entities.

AlCalzone, 2023-03-09:
I'm not seeing anything there either. Are you controlling the device in Z-Wave JS UI or natively in Home Assistant? If its the latter, can you try Z-Wave JS UI, so we can maybe rule it out?

andlo, 2023-03-10:
I am dooing it from HA as I cant do it from Z-Wave JS UI

[image]

AlCalzone, 2023-03-10:
Tr re-interviewing in zwave js UI. Looks like the binary switch cc version is unknown, so Likely an interview timeout.

andlo, 2023-03-10:
Tryed that, and also tryed activate Increase node report timeut, but result is the same :(
Would a new log show something ?

AlCalzone, 2023-03-10:
Maybe, this time interview and capture the log in zwave-js UI please.

andlo, 2023-03-10:
Ill reintervieved and made a new log with loglevel debug 

I really appreciate your help :) 

[zwavejs_2023-03-10.log](https://github.com/zwave-js/node-zwave-js/files/10942376/zwavejs_2023-03-10.log)

AlCalzone, 2023-03-10:
How does it look now in Z-Wave JS UI? Which controls are you missing?
The interview looked okay.

andlo, 2023-03-11:
I would say I am missing the binary switches 
[image]

AlCalzone, 2023-03-14:
I think I might have it, you will have to test next release though.

closed 2023-03-14 by zwave-js-bot