zwave-js/zwave-js #5549: NVM Restore from Aeotec zw090 (Gen 5, firmware 1.02) to Zooz zst10 (Gen 7, firmware 7.18.3) results in all nodes dead

Hooverjc, 2023-03-09:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
NVM Restore from Aeotec zw090 (Gen 5, firmware 1.02) to Zooz zst10 (Gen 7, firmware 7.18.3)

What do you observe?
All nodes are dead. Lots of TX and RX going on, but in PC Controller software, I see lots of "Home ID" errors.

What did you expect to happen?
I expected all the nodes on my Aeotec Gen5 to be functional on my Zooz Gen7.

Steps to reproduce the behavior:

1. Backed up my Aeotec Gen5 with Z-Wave JS UI
2. Restore the backup of my Gen5 on my Gen7 with Z-Wave JS UI
3. Restart

### Device information

Old controller : Aeotec Gen5 SDK 6.7
New controller : Aeotec Gen7 SDK 7.18.1
Home automation system : Home Assistant with zwave-js-ui docker running latest release
zwave-js-ui: 8.10.0.8737258
zwave-js: 10.11.0

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
zwave-js-ui: 8.10.0.8737258
zwave-js: 10.11.0

### Did you change anything?

yes (please describe)

### If yes, what did you change?

I mean, I'm trying to migrate to a different zwave 700 stick as my old aeotec gen7 was a hot mess. I went back to this gen 5 stick and most of my issues went away, but now I am testing a few other 700 sticks to see if the results are consistent or due to the poor antenna design in the aeotec gen 7.

I was able to NVM backup/restore from Aeotec Gen 7 to Aeotec Gen 5 without issue. Now Aeotec Gen 5 to Zooz Gen 7 is being wonky.

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwavejs_current.log](https://github.com/zwave-js/node-zwave-js/files/10934365/zwavejs_current.log)

If you need me to do any different logging, provide the BIN file backup, or anything liek that, just let me know.

AlCalzone, 2023-03-09:
Did you check that the RF region is still correct?

Hooverjc, 2023-03-09:
I did in both the UI and PC Controller software. Just to be sure, I just factory reset the stick, took a fresh backup, and reapplied it:

[image]

Hooverjc, 2023-03-09:
also, here is the complete debug log from the latest run including generation of the NVM backup, restore, and subsequent issues. Its kinda large, but hoping it will have any relevant info you need in it.
[zwave.log](https://github.com/zwave-js/node-zwave-js/files/10936088/zwave.log)

AlCalzone, 2023-03-09:
That foreign home ID thing looks strange. Can you send me the Gen5 NVM backup and a new backup from the Zooz stick via e-mail (see my profile)? I'd like to take a look at the files.

Hooverjc, 2023-03-09:
sure can, will have this over to you shortly

Hooverjc, 2023-03-09:
> That foreign home ID thing looks strange. Can you send me the Gen5 NVM backup and a new backup from the Zooz stick via e-mail (see my profile)? I'd like to take a look at the files.

I re-read this and realized you maybe wanted a clean backup (factory reset) for that zooz stick. Let me do that for you as well so you have something clean to reference just in case.

AlCalzone, 2023-03-09:
The one after migration is more important, so if you do that (can't hurt), migrate again afterwards and send me a backup of that too.

Hooverjc, 2023-03-09:
Cool, that's the gen7_backup.bin in the last email, so you should have both now

AlCalzone, 2023-03-14:
I took a look at your files. Not sure how this happened, but your stick has two contradicting versions stored in its memory. From what I've gathered from the Gecko SDK sources, the **application version** (i.e. firmware version) should be updated every time the stick boots and notices that the stored version is less than the one currently being executed. If necessary this will also trigger some migrations of the stored data.
The **protocol version** seems to only change when the format of the protocol memory also changes, which means `protocolVersion <= applicationVersion`.

As a result, Z-Wave JS uses the application version to determine the format of the data it writes and deduces the protocol version from that. This makes it possible to downgrade the NVM to an older version, which the sticks cannot do themselves.

However, it looks like your stick updated the protocol version instead of the application version on update. Not sure how that can happen when the firmware is supposed to be the same. As a result it has protocol version `7.18.3` (which is after a format change) and application version `7.15.1` (which is before that format change).
This means it cannot read the format Z-Wave JS wrote, causing it to give itself a new home ID on startup.

Anyways, I've sent you a new NVM file where I fixed these versions manually. Unfortunately, you won't be able to restore it as-is since Z-Wave JS will attempt a conversion due to the version mixup I mentioned above.
I'll figure out with @robertsLando how you can skip that conversion step.

robertsLando, 2023-03-14:
@Hooverjc Just put the nvm backup file in store directory then open advanced dialog and select "Driver function". Now paste inside the code input this:

```
const { logger, zwaveClient, require } = this
const fs = require('fs')
const { join } = require('path')
const { storeDir } = require('../config/app')
// change file name here
const buffer = fs.readFileSync(join(storeDir, 'nvmbackup.bck'))

await driver.controller.restoreNVMRaw(buffer, (bytesWritten, total) => {
  logger.info('NVM Backup: ' + Math.round(bytesWritten / total*100))
})
```

Keep logs window opened to see if there are errors and remember to replace `nvmbackup.bck` with your nvm backup file name

AlCalzone, 2023-03-14:
(updated the explanation to add a few details)

Hooverjc, 2023-03-14:
Was able to test this out and it did end up correcting the issue and all of my nodes are functioning correctly on the new 700 stick (time will tell if its any better than the aeotec gen 7 stick, but so far no issues). Will I have to do anything special in the interim with a firmware update or anything like that to avoid this issue in the future or is this a one-time fix that should be good on this specific device going forward?

Thank you very much for your assistance on this!

AlCalzone, 2023-03-14:
This should keep working now that everything is in order. Migrating from the Gen5 again may need some manual intervention though.

closed 2023-03-14 by AlCalzone