zwave-js/zwave-js #5564: Z-wave User Lock Code Manager

skl111, 2023-03-13:
**Is your feature request related to a problem? Please describe.**
An Official Z-wave user Lock Code manager

**Describe the solution you'd like**
A clean implementation of a Z-Wave Lock Code Manager, maybe integrate it into a larger Lock Code Manager.

**Describe alternatives you've considered**
I installed and used Keymaster when I was testing out HA. It was not user friendly.

**Additional context**
HomeSeer has this feature, and it works pretty well. I experimented to HA while HS was having issues with Google Home because HA worked with it, and User Lock Code management not having and officially supported user friendly option was a show stopper for me, which I hated because there were so many other features I liked and I got it working with the HomeSeer Zwave card!

I have attached a screenshot of the HomeSeer User Lock code management screen to show the ease of use of it.

[
[image]
](url)

jmgiaever, 2023-03-13:
Hi,

I think this is something that you should request with HA? 

Or with Zwave-js-ui that is the most popular interface for Zwave-js.

AlCalzone, 2023-03-13:
This 👆️