zwave-js/zwave-js #5588: ZMNHAD - Missing class 48

biggranben, 2023-03-17:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Class 48 Instance 1, 2 and 3 are missing

What do you observe?
Can't access to the state of the input I1, I2, I3

### Device information

Manufacturer: Qubino
Model name: ZMNHAD
Node ID in your network: 64
Device ID : 345-82-2 (0x0159-0x0002-0x0052)

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
zwave-js-ui: 8.6.1.7833cd8
zwave-js: 10.3.1

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Working on OpenZWAVE

[image]

### Attach Driver Logfile

[zwave_17-03-2023.log](https://github.com/zwave-js/node-zwave-js/files/10998189/zwave_17-03-2023.log)

biggranben, 2023-03-17:
[z-ui_2023-03-17.log](https://github.com/zwave-js/node-zwave-js/files/10998209/z-ui_2023-03-17.log)

AlCalzone, 2023-03-17:
That's also not a driver log. Please read the link the bot posted again.

biggranben, 2023-03-17:
[zwave_17-03-2023.log](https://github.com/zwave-js/node-zwave-js/files/11003926/zwave_17-03-2023.log)

This one is correct ?

biggranben, 2023-03-26:
Do you need anything else ?

AlCalzone, 2023-03-26:
still not the correct one. The correct one is called `zwavejs_<date>.log`

biggranben, 2023-03-26:
[zwavejs_2023-03-17.log](https://github.com/zwave-js/node-zwave-js/files/11073183/zwavejs_2023-03-17.log)

AlCalzone, 2023-04-03:
Not it's a driver log, but it does not include a re-interview of the problematic device like I asked [here](https://github.com/zwave-js/node-zwave-js/issues/5588#issuecomment-1473918310). Also make sure it's not filtered in any way.

biggranben, 2023-04-07:
[zwavejs_2023-04-07.log](https://github.com/zwave-js/node-zwave-js/files/11176260/zwavejs_2023-04-07.log)

I think now these one is correct ! Ok for you ?

AlCalzone, 2023-04-07:
According to that log:
* Endpoint 1 is a binary **switch**, which does not expose any sensor functionality.
* Endpoint 2 is a **multilevel** sensor, not a binary sensor
* Endpoint 3 does not exist.

I'm assuming you skipped over this part in the manual:
[image]

Without doing this, the sensors won't show up, even if you changed the parameters.

closed 2023-04-07 by zwave-js-bot

biggranben, 2023-04-08:
Hello,

On this devise, there are 3 imput : 
- input 1 : linked to the relay
- input 2 : sensor
- input 3 : sensor
- +temperature sensor if connected to the specific port

[image]

Input 2 and 3 are configured as sensor binary : 
[image]
[image]

But no way to get the state.

It is a non documented class on the manual, but on the class 48, we could get the input state for each input (1, 2, 3). It was working on OpenZWAVE.

biggranben, 2023-04-08:
Working after reinclusion

PR #5640 (2023-04-07): fix(config): for Qubino ZMNHAD Relay, mention necessity for re-inclusion
fixes: #5588