zwave-js/zwave-js #5596: Fibaro Dimmer 2 nodes become dead

overas, 2023-03-24:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

YES, BUT a Z-Wave JS UI developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
The problem started when using Aeotec z-stick 7 as a zwave controller. I have tried using different firmwares 7.18.2, 7.19.1 and now 7.19.2, but the problem is the same. I have also tried different z-wave-js driver version from v.10.8.0, v10.10.0 and now I use v.10.12.0 which are bundled with z-wave-js-ui. I use now z-wave-js-ui 8.11.0. I have also tried to reinterview the node which help temporary, but after a while the node goes dead again.

What do you observe?
I have five Fibaro Dimmer 2 nodes (Product code: FGD212) which goes dead after a while. All of them have firmware v3.5. When I ping the node the device goes Alive again. I have other nodes which do not go dead (Heatit Z-TRM3, Aeotec Heavy Duty Switch, Fibaro Metered Wall Plug Switch, Aeotec MultiSensor 6). I have healed alle the nodes before this happen.

What did you expect to happen?
When I change the controller to Aeotec Z-Stick Gen5+ the Fibaro Dimmer 2 nodes are Alive and never goes dead.

I have opened a Home Assistent zwave bugreport yesterday which you can find here: https://github.com/home-assistant/core/issues/90195
In this bugreport you can find logs

### Device information

Manufacturer:  Fibaro
Model name:  Fibaro Dimmer 2 (Product code: FGD212) 
Node ID in your network: 13, 14, 16, 27, 43

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 10.12.0
`node-zwave-js` branch: 
`zwave-js-ui` branch: 8.11.0

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

When I change the controller to Aeotec Z-Stick Gen5+ the Fibaro Dimmer 2 nodes are Alive and never goes dead.

### Attach Driver Logfile

I have opened a Home Assistent zwave bugreport yesterday which you can find here: https://github.com/home-assistant/core/issues/90195
In this bugreport I have attached logs which I hope can help find out this problem. If you need more logs I am glad to help.

AlCalzone, 2023-03-24:
What the bot said

overas, 2023-03-24:
Here is the logfiles you need. Node 13 and 14 was going dead short time after I started up zwave-js-ui. I tried to reconfigure node 13 and 14, but node 13 and 16 was going dead short time after. I use a USB extension cable between z-stick7 and the Intel nuc. 

Node 13 Taklys stue1etg rad1 export json:
[node_13.zip](https://github.com/zwave-js/node-zwave-js/files/11064520/node_13.zip)

Zwave-js logs:
[zwave-js-ui-store.zip](https://github.com/zwave-js/node-zwave-js/files/11064538/zwave-js-ui-store.zip)

AlCalzone, 2023-04-04:
I don't see anything obviously wrong with your configuration - no excessive meter reports or anything.

Except you're using the EU version of Z-Stick 7. Not sure what the problem with that one is exactly, but we no longer recommend using it. Apparently the 7.19.2 firmware didn't help. There are countless threads about its subpar RF performance, e.g. https://github.com/zwave-js/node-zwave-js/issues/4352

A lot of the incoming RSSI measurements are around -80...-100 dBm, which isn't much above the background noise level of -110 dBm. As a result, some ACKs seem to get lost, which sometimes triggers nodes to re-transmit their messages a few times - and that is enough to make the stick stop transmitting at all.

I'm afraid there's not much to be done, except make an NVM backup, get a different stick and restore the NVM there. Possible alternatives:
- Aeotec Gen5+ (the + is important), with firmware 1.2 (update is [here](https://aeotec.freshdesk.com/support/solutions/articles/6000252294-z-stick-gen5-v1-02-firmware-update) if necessary)
- Zooz ZSt10-700 (if you can get your hands on a EU version)
- Silicon labs UZB7 (or UZB3 works too, but that's also the older 500 series)
- 800 series does not support NVM restore yet, last I tested.

closed 2023-04-04 by AlCalzone

ceesvanegmond, 2023-10-16:
@AlCalzone I've done exactly this. Had the same errors as OP and switched from my Aeotech 700 stick to a Aeotech Gen5+ stick. On the 700 series I did an NVM backup (from Home Assistant ZwaveJS ui) and restored it on my Gen5. The restore was successful and I can see al my Z-Wave devices. However actually having the same sort of issues. I tried setting the power levels as you mentioned earlier but getting errors doing that and the USB devices doensn't seem available anymore executing that in the Driver section. 

Still having these errors (a lot)
```
2023-10-16T17:09:42.869Z DRIVER « [Node 169] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Duplicate command (sequence number 137)
```

I am on the Latest Gen5+ firmware (v1.2).  But I don't know how to debug this further.