zwave-js/zwave-js #5599: MH10-PM2.5-WA don't report pm2.5 value

athamour1, 2023-03-28:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

  The device either don't send via Z-Wave the pm2.5 value, or it is missing from the configuration.

What do you observe?

  The pm2.5 value isn't available in "Multilevel Sensor v5" section.

What did you expect to happen?

  The pm2.5 value to appear 

Steps to reproduce the behavior:
  1. Add the device in the z-wave network
  2. Observe the "Multilevel Sensor v5" section of the device.

### Device information

Manufacturer: MCO-Home
Model name: MH10-PM2.5-WA
Node ID in your network: 5

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
zwave-js-ui: 8.11.0.598632a
zwave-js: 10.12.0

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

```
2023-03-28 12:13:41.002 INFO APP: GET /control-panel 304 33.457 ms - -
2023-03-28 12:13:41.058 INFO APP: GET /static/css/app.14b2d5b1f7d79cd90cd7.css 304 11.284 ms - -
2023-03-28 12:13:41.066 INFO APP: GET /static/js/126.2000332916bde630c84b.js 304 4.338 ms - -
2023-03-28 12:13:41.070 INFO APP: GET /static/js/app.d3b4f1244092d962adaa.js 304 6.472 ms - -
2023-03-28 12:13:41.387 INFO APP: User admin logged in successfully from 192.168.1.17
2023-03-28 12:13:41.415 INFO APP: POST /api/authenticate 200 40.020 ms - 221
2023-03-28 12:13:41.425 INFO APP: GET /api/auth-enabled 200 31.806 ms - 28
2023-03-28 12:13:41.464 INFO APP: GET /static/logo.png 304 1.319 ms - -
2023-03-28 12:13:41.588 INFO APP: GET //static/fonts/MaterialIcons-Regular.2d80174.woff2 304 1.243 ms - -
2023-03-28 12:13:41.647 DEBUG SOCKET: New connection Kz0Zdcw_lo82ZOiFAAAD
2023-03-28 12:13:41.688 DEBUG SOCKET: Event INITED emitted to Kz0Zdcw_lo82ZOiFAAAD
2023-03-28 12:13:41.737 INFO APP: GET /api/settings 304 117.413 ms - -
2023-03-28 12:13:41.859 INFO APP: GET /api/auth-enabled 304 4.700 ms - -
2023-03-28 12:13:50.368 DEBUG SOCKET: Event ZWAVE_API emitted to Kz0Zdcw_lo82ZOiFAAAD
2023-03-28 12:13:50.385 INFO Z-WAVE: Calling api refreshValues with args: [ 5, [length]: 1 ]
2023-03-28 12:13:50.468 INFO Z-WAVE: [Node 005] Metadata updated: 49-0-Air temperature
2023-03-28 12:13:50.471 INFO Z-WAVE: [Node 005] Value updated: 49-0-Air temperature 19.2 => 19.1
2023-03-28 12:13:50.524 INFO Z-WAVE: [Node 005] Metadata updated: 49-0-Humidity
2023-03-28 12:13:50.527 INFO Z-WAVE: [Node 005] Value updated: 49-0-Humidity 39 => 38
2023-03-28 12:13:51.622 INFO Z-WAVE: Success zwave api call refreshValues undefined
```

AlCalzone, 2023-03-28:
Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` and attach it here as a file (drag & drop into the text field).

While the log is being recorded, re-interview the device (make sure to wake it up if it's a battery powered device).

athamour1, 2023-04-04:
@AlCalzone Is this the correct log?

AlCalzone, 2023-04-04:
It looks like the readings are being dropped because they are using a scale/unit that the device did not report as supported. Unfortunately, the log does not contain the re-interview I asked for, so I don't see what is supported.

athamour1, 2023-04-04:
I re-run the re-interview and this logs created 
[logs.zip](https://github.com/zwave-js/node-zwave-js/files/11147944/logs.zip)
@AlCalzone is saw inside now the interview logs

AlCalzone, 2023-04-04:
Yeah, there we have it. The device reports that it only supports the "Mole per cubic meter" unit (0) for PM2.5:
```
2023-04-04T11:23:18.619Z DRIVER « [Node 005] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -63 dBm
                                  └─[MultilevelSensorCCSupportedScaleReport]
                                      sensor type:      Particulate Matter 2.5
                                      supported scales:
                                      · Mole per cubic meter
```
but then uses the "Microgram per cubic meter" unit (1) for reports, even though "Mole per cubic meter" is requested. I suspect this is another classic case of a mis-formatted support bitmask, where `0000_0001` was meant to encode a 1 but actually means 0.

AlCalzone, 2023-04-04:
@zwave-js-bot add compat flag to `0x015f:0x0a02:0x5102`
```jsonc
// The device reports in a different scale than it reports support for
disableStrictMeasurementValidation: true
```

closed 2023-04-04 by AlCalzone

athamour1, 2023-04-04:
Thank you very much for the help @AlCalzone, much appreciated!

PR #5624 (2023-04-04): fix(config): disable scale validation for MCOHome MH10-PM2.5-WA/WD
fixes: #5599

**TODO:**
- [x]	Change PR title to be more specific