zwave-js/zwave-js #5600: Device parameters revert to "Configuration vundefined" in ZUI some time after a successful interview

kdknigga, 2023-03-28:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

YES, BUT a Z-Wave JS UI developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

I have a device, which after an interview looks correct.  The CCs I'd expect show up, etc.  Under "Configuration v4" I see the parameters I'd expect (being read out of a custom device config for now, as it's a brand-new device that hasn't been released yet), but no values for any of them.  If I click the "Refresh" button, the values are polled and displayed. 

Then time goes by... sometimes a couple of hours, sometimes a couple of days... and then I start seeing (on the same device as above) CCs "Configuration v4" and "Configuration vundefined".  Some params/values end up under v4 and some are in vundefined. 

Looking at my ZUI instance right now, it seems like all of the parameters are now in vundefined except for some that I update regularly via automation.  Those I touch regularly are in v4.

zwave-js-ui: 8.11.0
zwave-js: 10.12.0

### Device information

Manufacturer: Inovelli
Model name:  VZW31-SN
Node ID in your network: 90

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

Z-Wave JS UI HA Add-on: 1.8.0
zwave-js-ui: 8.11.0
zwave-js: 10.12.0

### Did you change anything?

yes (please describe)

### If yes, what did you change?

Custom device config file for an unreleased device.  It will be provided.

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

I'll send more information via email.

AlCalzone, 2023-04-11:
> Looking at my ZUI instance right now, it seems like all of the parameters are now in vundefined except for some that I update regularly via automation. Those I touch regularly are in v4.

If this is the case for the cache files you sent me, I think the problem actually lies in zwave-js-ui. There's exactly one location in the cache files referencing the Configuration CC version, and that is clearly a `4`:
```
{"k":"node.90.endpoint.0.commandClass.0x70","v":{"isSupported":true,"isControlled":false,"secure":true,"version":4}}
```

@robertsLando

AlCalzone, 2023-04-11:
I think this can only happen if `endpoint` is `undefined` here:
https://github.com/zwave-js/zwave-js-ui/blob/6d8e57de42270b002a612812cbf84d73e978381b/lib/ZwaveClient.ts#L4682-L4689

In the cache files, the values for Configuration CC are **only** on endpoint 0, which is the node itself and cannot be `undefined`.

Maybe this is some kind of race condition, where `_updateValueMetadata` is called before the node is ready, so `commandClassVersion` is never set, and it is never updated either.