zwave-js/zwave-js #5621: Honeywell T6 Pro TH6320ZW thermostat incorrectly reports Celsius without Preferred scales set

idatum, 2023-04-03:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Starting with Z-Wave driver version 10.12.0, server version 1.27.0, my Honeywell T6 Pro thermostat returns a Celsius value for [53-49-0-Air temperature] (air temperature) instead of Fahrenheit. I have [53-112-0-2] Temperature Scale set to Fahrenheit (value 0). Either hitting Refresh in Z-Wave JS UI Multilevel Sensor v5 or a reinterview causes the value to incorrectly switch. Also, it will periodically switch values on its own (no manual steps).

What do you observe?
I can rule out Home Assistant since both the Z-Wave JS logs (debug logs attached) and MQTT values show the incorrect C value.
The value will sometimes correctly show F, then incorrectly switch to C, and so no. A manual refresh or reinterview will always incorrectly show C.
I can temporarily fix by toggling temperature scale F --> C --> F.
Workaround: Add "Air temperature: Fahrenheit (°F)" to Preferred scales under Z-Wave Settings.

What did you expect to happen?
Z-Wave JS should always return Fahrenheit values for thermostats if temperature scale is explicitly set to Fahrenheit.
I should not have to set Preferred scales as a workaround.

Steps to reproduce the behavior:
  1. Go to Z-Wave JS UI Control Panel and note the correct Fahrenheit air temperature value displayed in Multilevel Sensor v5 for the T6 Pro Z-Wave Programmable Thermostat.
  2. Click Refresh in Multilevel Sensor v5 and the value changes to Celsius with F units.
  3. See incorrect value change in uploaded debug log.
  
[T6_thermostat_F_issue.log](https://github.com/zwave-js/node-zwave-js/files/11143120/T6_thermostat_F_issue.log)

### Device information

Manufacturer: Honeywell
Model name: T6 Pro Z-Wave Programmable Thermostat TH6320ZW
Node ID in your network: 53

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 10.12.0
`node-zwave-js` branch:
`zwave-js-ui` branch:

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

I previously did not need to set Preferred sacle to get the correct Fahrenheit air temperature value. This has worked for several versions (almost 2 years).

### Attach Driver Logfile

[zwavejs_current.log](https://github.com/zwave-js/node-zwave-js/files/11143185/zwavejs_current.log)

AlCalzone, 2023-04-04:
As far as I see it, this problem consists of two parts:
1. The thermostat supports two scales, Celsius and Fahrenheit. Without a preferred scale configured, the driver defaults to using the **first** supported scale for queries, which is Celsius. I suppose we could add another heuristic to re-use the most recent scale reported by the device for further queries.
2. When the report comes in, a `metadata updated` event is emitted, in which applications are informed about the the new scale of the value. You can see this (without the actual contents) in your log:
	```
	2023-04-03 15:44:29.886 INFO Z-WAVE: [Node 053] Metadata updated: 49-0-Air temperature
	2023-04-03 15:44:29.890 INFO Z-WAVE: [Node 053] Value updated: 49-0-Air temperature 21.5 => 21.5
	```
	Looks like HA does not evaluate that event correctly, so you get a mismatch between the value and the scale.

AlCalzone, 2023-04-04:
1. should be fixed by https://github.com/zwave-js/node-zwave-js/pull/5622
2. is https://github.com/zwave-js/zwave-js-ui/issues/3034

closed 2023-04-04 by zwave-js-bot