zwave-js/zwave-js #5625: Check Health - The node failed to decode the message. (ZW1405)

mkshb, 2023-04-04:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

YES, BUT a Z-Wave JS UI developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
The health check does not work for S2 authenticated nodes. 

What do you observe?
If I run a health check on a node without S2, it is completed successfully. With S2 nodes the error ZW1405 occurs. 

What did you expect to happen?
Health Check 

Steps to reproduce the behavior:
  1. Go to Network Graph
  2. Click on S2 Node and run Health Check against Controller.
  3. See error

```
2023-04-04T12:32:05.493Z CNTRLR   [Node 016] Starting lifeline health check (5 rounds)...
2023-04-04T12:32:05.493Z CNTRLR » [Node 016] requesting node neighbors...
2023-04-04T12:32:05.496Z SERIAL » 0x010700801001000069                                                 (9 bytes)
2023-04-04T12:32:05.496Z DRIVER » [REQ] [GetRoutingInfo]
                                    remove non-repeaters: true
                                    remove bad links:     false
2023-04-04T12:32:05.498Z SERIAL « [ACK]                                                                   (0x06)
2023-04-04T12:32:05.501Z SERIAL « 0x012001800066e3076880000000000000000000000000000000000000000000000 (34 bytes)
                                  034
2023-04-04T12:32:05.501Z SERIAL » [ACK]                                                                   (0x06)
2023-04-04T12:32:05.502Z DRIVER « [RES] [GetRoutingInfo]
                                    node ids: 10, 11, 14, 15, 17, 18, 22, 23, 24, 25, 26, 27, 36, 38, 39, 48
2023-04-04T12:32:05.503Z CNTRLR « [Node 016] node neighbors received: 10, 11, 14, 15, 17, 18, 22, 23, 24, 25, 26
                                  , 27, 36, 38, 39, 48
2023-04-04T12:32:05.507Z SERIAL » 0x0108001310010025ba6a                                              (10 bytes)
2023-04-04T12:32:05.507Z DRIVER » [Node 016] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      186
                                  └─[NoOperationCC]
2023-04-04T12:32:05.508Z SERIAL « [ACK]                                                                   (0x06)
2023-04-04T12:32:05.514Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2023-04-04T12:32:05.515Z SERIAL » [ACK]                                                                   (0x06)
2023-04-04T12:32:05.515Z DRIVER « [RES] [SendData]
                                    was sent: true
2023-04-04T12:32:05.534Z SERIAL « 0x01180013ba00000200bc7f7f7f7f0000030000000003010000f1              (26 bytes)
2023-04-04T12:32:05.534Z SERIAL » [ACK]                                                                   (0x06)
2023-04-04T12:32:05.535Z DRIVER « [REQ] [SendData]
                                    callback id:            186
                                    transmit status:        OK, took 20 ms
                                    routing attempts:       1
                                    protocol & route speed: Z-Wave, 100 kbit/s
                                    ACK RSSI:               -68 dBm
                                    ACK channel no.:        0
                                    TX channel no.:         0
2023-04-04T12:32:05.548Z SERIAL » 0x0108001310010025bb6b                                              (10 bytes)
2023-04-04T12:32:05.549Z DRIVER » [Node 016] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      187
                                  └─[NoOperationCC]
[... 389 lines cut ...]
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwave-js-ui: 8.12.0.c982de8
zwave-js: 10.12.0

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### If yes, where did it work?

I can't say. Network is newly built and sometimes reacts very hesitantly.

### Attach Driver Logfile

[zwavejs_current.log](https://github.com/zwave-js/node-zwave-js/files/11148659/zwavejs_current.log)

AlCalzone, 2023-04-04:
FWIW, I just completed a health check on an S2 authenticated node without issues. So this is something specific to your setup and requires a log to even begin troubleshooting.

mkshb, 2023-04-04:
Hi there,

sorry about the form. I had filled out everything and probably misclicked somewhere.
Here is the rest of the info.

Which branches or versions?
zwave-js-ui: 8.12.0.c982de8
zwave-js: 10.12.0

Did you change anything?
No, but it's a fresh build network.

Did this work before?
Some devices respond with a time delay, but not always. 

Attach Driver Logfile
[zwavejs_current.log](https://github.com/zwave-js/node-zwave-js/files/11148944/zwavejs_current.log)

closed 2023-04-04 by zwave-js-bot

mkshb, 2023-04-06:
Just a little feedback and a big thank you!
Works with the current master Docker image.