zwave-js/zwave-js #5626: Issue with Zipato mini keypad rfdi (wintop)

cokeman0, 2023-04-04:
Hi,

After upgrading from zwavejs2mqtt 1.7.2 to 1.8.0 / 1.8.1 and 1.9.0

my 2 wintop keypads are acting up.

in the old 1.7.2, when entering og leaving, press home or away and hold up the rfdi tag.

in 1.8.x it's the same, BUT under 1.8.X it's like the wintop keypads wakes up, and resends the last action.

so when leaving the home, we are the alarm, but if the second keypad was last used to disarm, the the alarm are turned off and on durring the day

then event in HA (listen to events)  

there is no difference the notification in HA of a real event or the "resend"

The "resend" happens with the "wakeup", if that helps.

This was NOT an issue before 1.8.x and I had it running for at least 2.5 years

cokeman0, 2023-04-04:
ID 43 reinterviwed

I hope you can find so errors in that log

[zwave-js-ui-store (1).zip](https://github.com/zwave-js/node-zwave-js/files/11148910/zwave-js-ui-store.1.zip)

AlCalzone, 2023-04-04:
Yeah, I know what's wrong:
A recent update implemented a requirement that some nodes/CCs need to be polled every couple of hours. For Notification CC that is only a requirement when the node operates in "pull" mode
Your device is incorrectly detected as operating in "pull" mode, because of an edge case in the mode discovery. As a result it gets queried roughly every 6h when it wakes up and it seems to return the last notification it sent.

cokeman0, 2023-04-04:
I do hope, that can be fixed 😄 but, yes, that sounds about right, my devices wakes up every 2 hours, and then the 6 hours pull, makes the strange pattern I can see in HA

AlCalzone, 2023-04-04:
Fix is up at https://github.com/zwave-js/node-zwave-js/pull/5627
Once that is released, you will have to re-interview the devices again for the change to take effect.

cokeman0, 2023-04-04:
Great... Not sure how to update, as I'm running the HA version...

closed 2023-04-04 by zwave-js-bot