zwave-js/zwave-js #5638: Battery based devices don't get marked as dead if they haven't checked in recently

Hooverjc, 2023-04-07:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
When a battery based device fails to check-in due to a dead battery, this device continues to be marked as "assleep" despite the fact that it will never again check-in. Because the battery indicator won't update any longer, this can cause battery based devices to go dead unnoticed.

What do you observe?
The device remains "asleep" despite it having a dead battery. I've seen this go on for weeks. Usually i'd catch the battery getting low and set warnings, but in the winter, Lithium batteries can die very quickly, causing a device to go from 50+% to 0 in a few hours. This results in a dead device with a previously reported high battery level and no automatic form of flagging the device as dead.

What did you expect to happen?
I'd expect some sort of configurable timeout. For example, if a device is battery based and doesn't check-in at all for X days, perhaps either marking it as dead or raising some sort of event notifying that the device may be unresponsive. Not sure if this would be global or per device, but some option here to bring light to these situations in an automatic fashion would be nice.

Steps to reproduce the behavior:
  1. Add a battery device and make sure its configured and marked as active with all parameters populated.
  2. Remove the devices battery
  3. Wait for a week or more and the device will continue to be marked as "asleep".

### Device information

Seems agnostic to devices as all my battery devices can be impacted
Manufacturer: Zooz
Model name: ZSE41
Node ID in your network: 46

This particular node has been dead for weeks, yet is still marked as asleep. Device specific settings are configured to wake up every 12 hours:

[image]

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwave-js-ui: 8.13.0.02c66b0
zwave-js: 10.13.0

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

I can attach a log if helpful, but i'm not sure if there will be much value as the node isn't checking in and I don't see any activity in the logs for this node which is likely part of the problem.

blhoward2, 2023-04-08:
This is by design. We have no way of knowing how long a battery device may go. I have a water sensor that is set not to check-in and it can go a year or more before calling anything in.

AlCalzone, 2023-04-18:
Duplicate of #4207

closed 2023-04-18 by AlCalzone