zwave-js/zwave-js #5641: Erroneous/faulty sensor reports

jkaberg, 2023-04-08:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Bad network? Bad sensor?  I tried re-interviewing the devices several times. While both the devices in this screenshot is of the same device, I've seen this behavior across many different devices. My theory is that its due to either network congestion or just bad signal?

What do you observe?
Sometimes sensors just give of bad readings like these
[image]
[image]

What did you expect to happen?
Good readings 😄 Perhaps Zwave-js can detect and (optionally?) discard these bad readings. Are there any checksum or so one could make use of?

### Device information

Manufacturer: Heatit
Model name: Z-trm2fx 

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: main/master
`zwave-js-ui` branch: main/master

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

As this behavior is quite sporadic its hard to get logs of it. But I'll try

andersevenrud, 2023-04-08:
Firstly, have you checked if there are any firmware updates available for your device ?

> Sometimes sensors just give of bad readings like these

> Perhaps Zwave-js can detect and (optionally?) discard these bad readings. Are there any checksum or so one could make use of?

Do other sensors outside kWh exhibit this behaviour ? If so you might be running into the same issue that I've been very frustrated with.

Most of my `kWh` readings from my switches are bad (across different brands). The value reports and checksums actually are correct... my devices are just bugged (zeroes, negative numbers like #2692, and just generally overshooting calculations up to +50%, etc.).

In my case I just gave up using the reported metering values and set up my own based on the `W` sensors, which seems very accurate using [Riemann sum integration](https://www.home-assistant.io/integrations/integration/) and [Utility Meter](https://www.home-assistant.io/integrations/utility_meter/) in HASS.

jkaberg, 2023-04-08:
@andersevenrud Yeah newest available firmware on those, and AFAIK on the rest of my rig as well. 

What you describe regarding issues with sensors is very similar to what I'm seeing, pretty much the same. I might very well do what you suggest as this is quite frustrating (having to adjust bad readings)

I'll let this stew for a few days so the maintainers see it and know its out there.

andersevenrud, 2023-04-08:
@jkaberg 

> What you describe regarding issues with sensors is very similar to what I'm seeing, pretty much the same

I suspected as much, especially when I saw which brand it was. It's one that's sold at the distributor where I got my bugged switches from (one of which I've been [documenting](https://gist.github.com/andersevenrud/5d13d30bfcc08a4e121222671f8cb9e9#file-1_zpr-111-md))... and I have a suspicion that a lot of this stuff are white-label devices, which is why there's so many similar (or even *identical*) reports out there in the wild.

Seems like some of the gateways/bridges you can buy actually has some rules that drops (and/or corrects) some of these bugged values (I used this before going over to my own dongle and HASS). But the overshooting meters is still an issue there... so there's definitively some kind of fault here.

Hope you get it sorted out.

God helg! :)

jkaberg, 2023-04-09:
I just read @AlCalzone reply [here](https://github.com/zwave-js/node-zwave-js/issues/2692#issuecomment-1012907609) and while I perfectly agreed on his sentiment I'm afraid in the short term not much will happen. 

I was thinking of an possible (but it stinks) solution would be for HASS to have some form of "ingestion filter" where before consuming new (sensor) data it would be possible to apply filter (like low-pass, outlier etc) per sensor/entity. Just an idea... 😄 The current "fix/approach" with template sensors/filters just ends up in duplicated data.

With the reply in mind, I will close the issue

closed 2023-04-09 by jkaberg

jkaberg, 2023-04-09:
@andersevenrud made an [feature request](https://community.home-assistant.io/t/ingestion-filters-for-sensors/557948) over at the HASS forums to continue the discussion over there.