zwave-js/zwave-js #5643: Danfoss living thermostat LC-13  refusing the interview after it already worked.

linuxstony, 2023-04-09:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

The thermostat has been working before but refused to cooperate with the "better thermostat" integration in Homeassistant. So I thought - lets remove it and add it again - maybe it will work then. Sadly the removal did work, but the readdition did not. 

What do you observe?

The thermostat is factory reset and it is removed in Z-Wave JS Ui. When I try to add it again the interview fails 5/5 times and then aborts the process. 

What did you expect to happen?

I would like it to be interviewed correctly :+1: 

Steps to reproduce the behavior:
  
Normal adding the device procedure :(

### Device information

Manufacturer:  Danfoss Living Thermostat
Model name: LC13
Node ID in your network: 009

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwave-js-ui: 8.13.0.02c66b0
zwave-js: 10.13.0
home id: 3558914921
home hex: 0xd420bb69

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Same version and the version before - sadly i dont know the version anymore. 

### Attach Driver Logfile

2023-04-09 19:07:15.724 INFO Z-WAVE: [Node 009] Added with security None
2023-04-09 19:07:30.055 ERROR Z-WAVE: [Node 009] Interview FAILED: Attempt 1/5 failed
2023-04-09 19:07:30.058 DEBUG MQTT: Publishing to zwave/nodeID_9/status: { time: 1681067250057, value: true, status: 'Awake', nodeId: 9 } with options { qos: 1, retain: true }
2023-04-09 19:07:49.537 ERROR Z-WAVE: [Node 009] Interview FAILED: Attempt 2/5 failed
2023-04-09 19:07:49.541 DEBUG MQTT: Publishing to zwave/nodeID_9/status: { time: 1681067269540, value: true, status: 'Awake', nodeId: 9 } with options { qos: 1, retain: true }
2023-04-09 19:08:13.952 ERROR Z-WAVE: [Node 009] Interview FAILED: Attempt 3/5 failed
2023-04-09 19:08:13.955 DEBUG MQTT: Publishing to zwave/nodeID_9/status: { time: 1681067293954, value: true, status: 'Awake', nodeId: 9 } with options { qos: 1, retain: true }
2023-04-09 19:08:43.397 ERROR Z-WAVE: [Node 009] Interview FAILED: Attempt 4/5 failed
2023-04-09 19:08:43.398 DEBUG MQTT: Publishing to zwave/nodeID_9/status: { time: 1681067323398, value: true, status: 'Awake', nodeId: 9 } with options { qos: 1, retain: true }
2023-04-09 19:09:17.800 ERROR Z-WAVE: [Node 009] Interview FAILED: Maximum interview attempts reached
2023-04-09 19:09:17.805 DEBUG MQTT: Publishing to zwave/nodeID_9/status: { time: 1681067357803, value: true, status: 'Awake', nodeId: 9 } with options { qos: 1, retain: true }

AlCalzone, 2023-04-11:
What the bot said

linuxstony, 2023-04-11:
[zwavejs_2023-04-10.log](https://github.com/zwave-js/node-zwave-js/files/11199334/zwavejs_2023-04-10.log)
[zwavejs_2023-04-11.log](https://github.com/zwave-js/node-zwave-js/files/11199336/zwavejs_2023-04-11.log)

Sorry - wasnt able to upload yesterday with the Dolphin FileManager from KDE. Attached the logs from yesterday and today.

Hope those help.

AlCalzone, 2023-04-11:
Looks like the device was assumed to be awake when you restarted the interview, and we don't correctly update the assumed state when GetNodeInfo fails (due to the device not actually being awake).

closed 2023-04-11 by zwave-js-bot

linuxstony, 2023-04-11:
On the initial interview the thermostat was awake (the normal pairing procesdure with these kind of thermostats) - the interview failed nevertheless. Sadly there was no different outcome - the interview never completes now.

linuxstony, 2023-04-11:
[image]

AlCalzone, 2023-04-11:
If even the initial interview fails due to this issue, there's something wrong with the communication between the stick and the thermostat on a lower level.
I tested the PR with my LC-13 and it fixed the issue for me.

linuxstony, 2023-04-11:
I really thank you for your time and effort !! I will test again then and report back either ways.  

Again - thanks and have a nice day.