zwave-js/zwave-js #5644: Some Devices in Provisioning List are not being included, with error "NWI Home ID not found"

tsx-5, 2023-04-10:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

YES, BUT a Z-Wave JS UI developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
- Scan device QR Code to add DSK to Provisioning List.  Power up device and check to see if added.
What do you observe?
- Some subset of devices are not automatically added, after being added to the provisioning list
What did you expect to happen?
- All devices to be automatically added, after being added to the provisioning list
Steps to reproduce the behavior:
  1. Scan a device -- ones I've seen have the most failures are from Zooz and HomeSeer.
  2. Power on or Air Gap device
  3. Observe device not being added
  4. See error: NWI Home ID not found in provisioning list, ignoring request

### Device information

Manufacturer:  Zooz
Model name: ZSE41, ZSE11

Manufacturer: HomeSeer Technologies
Model number: HS-WD200+	

<br class="Apple-interchange-newline">

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:
`zwave-js-ui` branch:

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

[zwave-js-ui-store.zip](https://github.com/zwave-js/node-zwave-js/files/11190200/zwave-js-ui-store.zip)

tsx-5, 2023-04-10:
The log file is in the zipped file.

tsx-5, 2023-04-11:
[provisioningEntries (copy).txt](https://github.com/zwave-js/node-zwave-js/files/11205783/provisioningEntries.copy.txt)

AlCalzone, 2023-04-12:
I compared the corresponding NWI home IDs with what's in your logfile. There are none that come even close to the transmitted NWI home ID. If you know which two devices those are, I can take another look at the QR code.

I suspect that the QR codes got mixed up somehow and don't match the data the device sends.

tsx-5, 2023-04-15:
[zwavejs_2023-04-15.log](https://github.com/zwave-js/node-zwave-js/files/11239272/zwavejs_2023-04-15.log)

tsx-5, 2023-04-15:
If you could take a look at this latest logfile - the previous one seemed to only list one device (which I haven't figured out what device it is).  The one I just added should include other devices...  Including some that were included (on this very system) before.  I've attached a screenshot - the thing I would want you to take away is that a number of devices show "unknown" for some part of an entry.  Last night every one of those entries was filled in and accurate.
[image]

AlCalzone, 2023-04-17:
Are all of those devices actually included without Security? I would assume that they are using S2 if you included them via SmartStart.

The logfile is empty btw.

tsx-5, 2023-04-17:
[zwavejs_2023-04-15.log](https://github.com/zwave-js/node-zwave-js/files/11249367/zwavejs_2023-04-15.log)

tsx-5, 2023-04-17:
Uploaded logfile again - not sure what happened.  Most of those device were included with security at one point.  Then something happened and all of them lost correct inclusion (and operation).  Some of these devices include the Zooz devices, which is notable only in that I'm using a Zooz dongle.

AlCalzone, 2023-04-17:
You have a lot of this in your log:
```
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: No security class granted
```

Almost looks like you lost/broke the cache files and/or configuration.

tsx-5, 2023-04-17:
How would I try and correct that?  I don't think I did anything intentionally to mess with cache or configuration files.

AlCalzone, 2023-04-17:
Check the Z-Wave section in the addon settings - all the keys should be provided there.
If yes, re-interview all problematic devices 1-by-1 and check the checkbox for "reset security classes".

If not, I hope you have a backup. Else you'll need to exclude and re-include all devices that use Security.

tsx-5, 2023-04-17:
So,
All the keys are there - and I did nothing to change them.  However, re-interviewing does not address the issue.  It had once before - a week ago things fell a part, similar to what we have now.  but doing the re-interview (resetting security) set everything back to normal.  No such luck now.

AlCalzone, 2023-04-17:
I can look at a log of such a re-interview if you want.

tsx-5, 2023-04-17:
[zwavejs_2023-04-17.log](https://github.com/zwave-js/node-zwave-js/files/11250852/zwavejs_2023-04-17.log)

AlCalzone, 2023-04-17:
The interviews are marked as done immediately after the list of CCs is received. Seems to me like you didn't check the "reset security classes" checkbox?

tsx-5, 2023-04-17:
I did make sure to check the "reset security classes" checkbox.  Especially on my last two attempts (should be towards bottom of log file)

AlCalzone, 2023-04-17:
I mean those, yeah. Let's try loglevel "silly" - maybe that contains more information.

tsx-5, 2023-04-17:
[zwavejs_2023-04-17.log](https://github.com/zwave-js/node-zwave-js/files/11251119/zwavejs_2023-04-17.log)

Ok - the wackiness continues...  I went and changed the logging level to "silly" and then went back to the main UI screen with intent to rerun an re-interview.  But, what did I see?  My device titles / manufacturer were back (out of the blue).  I then went a head and run the interview (a couple of times) and for the ones I ran, security came back.  Log is included for all this weridness.

AlCalzone, 2023-04-17:
Looks better already. Not sure what the heck that was though. It seems that you have some connectivity issues though, during the interview Nodes 7 and 9 start repeating their messages a couple of times.

closed 2023-05-07 by zwave-js-assistant[bot]