zwave-js/zwave-js #5650: Missing device configuration: <enter manufacturer and device name>

I1100, 2023-04-12:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Yale YRD450-ZW3

### Manufacturer ID

0x0129

### Product Type

0x8104

### Product ID

0x45D5

### Firmware Version

2.32.0

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/4794

### Do you have a manual?

https://products.z-wavealliance.org/ProductManual/File?folder=&filename=product_documents/4751/YRC612%20Manual.pdf

AlCalzone, 2023-04-12:
@zwave-js-bot import config 4794 from zwa

AlCalzone, 2023-04-12:
That manual is for a bunch of variants and explains which one has which IDs and which config parameters - which is really good IMO. I think we should go through it and apply all of the information in there to add or update files.

closed 2023-08-08 by zwave-js-bot