zwave-js/zwave-js #5651: Device adds non existing capabilities

dxfoxd, 2023-04-12:
### Checklist

- [x]  I am not using Home Assistant. Or: a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is not described there.
- [x] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

zwave-js-ui: 8.9.0.2d95700

### ZwaveJS version

zwave-js: 10.10.0 

### Describe the bug

I have 51 z-wave devices of different brands (Fibaro, Neo Coolcam, MCOhome Heiman and EcoDim. The controller is an Aeotec Z-Stick 7. 

I noticed that unwanted and non existing capabilities are added to devices at random. When re-interviewing the devices the extra capabilities are gone, but this means that I need to wake-up approximately half of 51 battery operated sensors. First I believed that device capabilities are somehow mixed up, but now also capabilities are added that don't exist on any device I use.(for example, I don't have any device that measures water consumption)

For example: 
Neo Coolcam door/window sensor, non-existing capabilities:
[39-50-0-value-196609] Water Consumption [Cubic meters] | 54.74 Cubic meters 
[39-50-0-value-66817] Electric Consumption [A] | 0.04 A

Fibaro Button, non-existing capabilities: 
[31-50-0-value-66817] Electric Consumption [A] | 0 A
[31-50-0-value-66049] Electric Consumption [W] | 0 W

Ecodim ECO-DIM Z-Wave Led dimmer, , non-existing capabilities: 
[53-49-0-Illuminance] Illuminance | 139 Lux

Neo Coolcam Light Switch 1 channel, non-existing capabilities:
[55-113-0-alarmType] Alarm Type | 0
[55-113-0-alarmLevel] Alarm Level | 0
[55-113-0-Access Control-Door state] Door state | [22] Window/door is open

Heiman CO Detector, non-existing capabilities
[35-50-0-value-66817] Electric Consumption [A] | 0.04 A
[35-50-0-value-66049] Electric Consumption [W] | 5.04 W

etc.

I'm lost here. Is this related to the controller, or to the zwave-js-ui or zwave-js integration? Or is it a defective controller? How can I check this?

### To Reproduce

Unknown

### Expected behavior

Not adding non-existing capabilities or as a around, pre populate device capabilities and disable interviewing.

### Additional context

I can add or send any log, please let me know what is needed to resolve this issue.

dxfoxd, 2023-04-12:
Curious for the fix!

On the other hand, I may expect such behavior of the brands like Neo Coolcam or MCOhome, but Fibaro and EcoDim? IMHO not likely.

AlCalzone, 2023-04-12:
This is nothing we can fix. This happens because the RF frames get corrupted on air, but in a way that the checksum still matches. See https://github.com/home-assistant/core/issues/77966#issuecomment-1449797663 for a more detailed explanation and some hints how this may be mitigated.
Z-Wave JS UI recently started exposing the background RSSI measurements as a graph, which you can use to compare the RSSI of your problematic nodes with the background RSSI. The difference between those two should be significant (> 20 dBm).

robertsLando, 2023-04-12:
> This happens because the RF frames get corrupted on air, but in a way that the checksum still matches

What the hell 😨

closed 2023-04-18 by AlCalzone