zwave-js/zwave-js #5682: Notification types interview fails for Sensative Strips (worked on early versions of Z-Wave JS)

mundschenk-at, 2023-04-18:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
A re-interview (for example after a firmware update) does not repopulate the notification types supported by the node.

What do you observe?
On re-interviewing, notification types are missing and only get added when the occur naturally. (This means that the entities in HA are also missing.)

The log shows that the interview for notification types fails.

What did you expect to happen?
Supported notification types should be discovered on interviewing a node. 

### Device information

Manufacturer: Sensative
Model name: Strips Guard 700
Node ID in your network: 107

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 10.15.0
`zwave-js-ui` branch: 8.14.0

### Did you change anything?

yes (please describe)

### If yes, what did you change?

Updated Z-Wave JS (via the Home Assistant Community Add-on)

### Did this work before?

Yes (please describe)

### If yes, where did it work?

When previously interviewed (probably several months ago).

### Attach Driver Logfile

[zwavejs_2023-04-18-interview-107.log.zip](https://github.com/zwave-js/node-zwave-js/files/11266025/zwavejs_2023-04-18-interview-107.log.zip)

AlCalzone, 2023-04-19:
This seems to be a duplicate of https://github.com/zwave-js/node-zwave-js/issues/5481
You should be able to work around this by temporarily enabling this setting in the Z-Wave group:
[image]

mundschenk-at, 2023-04-28:
Looks like the fix did not work (could be related to/caused by #5703). Should I provide logs?

AlCalzone, 2023-04-28:
Yes please

AlCalzone, 2023-05-15:
please retest with the latest version

mundschenk-at, 2023-05-15:
Sorry, got sidetracked waiting for a new UI release. Should I wait for v10.20.0 to be available the HA Z-Wave JS UI add-on or test with 10.17.1?

AlCalzone, 2023-05-15:
10.17.1 should be good.

PR #5683 (2023-04-19): fix: refresh CC response timeout for possibly matching partial responses
Until now, waiting for a response to a GET-type CC expected the final response to come within the timeout. However if Transport Service CC is involved (or even for long commands using the `reportsToFollow` flag), this timeout can often be too short.

We now test possible "partial" CCs if they are a potential match for an awaited CC. If they are, the corresponding timer will be reset.

fixes: #5682
fixes: #5481
closed 2023-05-15 by AlCalzone